Ultimaker/Cura #19814: Spiralize outer contour fails in 5.9.0-beta.1

catalintucureanu, 2024-10-26:
### Cura Version

5.9.0-beta.1

### Operating System

Linux

### Printer

Creality Ender 3v2

### Reproduction steps

Add a simple cylinder, activate Spiralize Outer Contour on a generic profile, slice. Preview sliced part.

### Actual results

Z seam is visible in preview with Spiralize Outer Contour activated. Actual printed part also shows z-seam. Same part sliced in 5.8.0 shows up fine (z-seams only on bottom layers and final layer).

[image]

### Expected results

No z-seam with Spiralize Outer Contour activated.

### Add your .zip and screenshots here ⬇️

[image]

GregValiant, 2024-10-26:
Thanks for the report.
I can duplicate this.
[image]

I wonder if "Scarf Z-seam" might have broken Spiralize?
@HellAholic here is one for you.

GregValiant, 2024-10-27:
Observations:
1. "Experimental" is the correct word and has been used properly.
2. The breakage list is turning into a laundry list.

GregValiant, 2024-10-31:
That was quick.  Thank you.

Ferrograph, 2024-12-18:
Im having difficulty with spiralise in 5.9. I was previously using 5.3.1 and spiraise in Normal mode would do the extra run around the top finish the spiral off nicely. In 5.9 it stops very briefly on the top where the preview shows start/stop just as in the images above, and puts a nasty pip in the print.

5.3.1 didn't do this and so in 5.9 this is broken for me. For my print its important that it makes no stops otherwise the print is ruined.

5.3.1:
[image]

Same settings in 5.9:
[image]

GregValiant, 2024-12-18:
@Ferrograph this report was closed and the original problem (that was the topic) appears to have been solved.
The issue you are reporting does seem to be occurring but it's a different problem.
Please open another bug report and include a project file.