Ultimaker/Cura #19823: Flickers like crazy, mouse cursor leaves "tracks"

php4fan, 2024-10-28:
### Cura Version

5.8.1

### Operating System

Manjaro Linux

### Printer

irrelevant

### Reproduction steps

1. open a STL file
2. slice it
3. preview
4. move the mouse cursor around, right-click and drag to rotate the scene

### Actual results

see the screencast

[stupid_cura.webm](https://github.com/user-attachments/assets/7f2a38b5-f2ad-4fde-8937-c56a2570cb22)

### Expected results

Shouldn't flicker; mouse cursor should not leave "prints"

### Add your .zip and screenshots here ⬇️

This field shouldn't be required, please fix it

LilBub, 2024-10-28:
Oh wow that is super weird.  They might want to know your system specs or cura log

php4fan, 2024-10-29:
I thought I had pasted this 🤦

Operating System: Manjaro Linux 
KDE Plasma Version: 6.1.5
KDE Frameworks Version: 6.6.0
Qt Version: 6.7.2
Kernel Version: 6.6.54-2-MANJARO (64-bit)
Graphics Platform: Wayland
Processors: 12 × 12th Gen Intel® Core™ i7-1255U
Memory: 15.3 GiB of RAM
Graphics Processor: Mesa Intel® Graphics
Manufacturer: ASUSTeK COMPUTER INC.
Product Name: Vivobook_ASUSLaptop X1502ZA_F1502ZA
System Version: 1.0

wawanbreton, 2024-11-07:
Hi @php4fan, can you test the 5.9.0 Beta (1 or 2) release ? We have applied a workaround for Cura on Wayland, it works correctly however that comes with lower 3D performances, especially visible in preview mode.