Ultimaker/Cura #19824: Support Blocker Failure

Annoyed123x, 2024-10-28:
### Cura Version

5.7.1

### Operating System

windows 11

### Printer

ulimaker s3

### Describe your model

bolt, only way to print without support on the threads is to completely turn off support

### Add your .zip here ⬇️

[image]

Annoyed123x, 2024-10-28:
Solved. Cura can't seem to use support blocker if you are printing from nozzle #2.

fieldOfView, 2024-10-28:
Check the "Support horizontal expansion" setting for your second nozzle. Set it to 0.

Annoyed123x, 2024-10-28:
Sorry, I went and got a different print running. I'm not sure I see that setting you are referring to. 
Both nozzles seem to have the same settings 
[image]
Update though, if I have both nozzles enabled it will generate correctly. If just nozzle 1 is enabled, it also works. But if I only have nozzle 2 enabled is when the error shows up. 
[image]
[image]

Annoyed123x, 2024-10-28:
That's the one. Didn't realize I needed to reassign the blocker also.

closed 2024-10-28 by Annoyed123x