Ultimaker/Cura #19833: (SCARF SEAMS) Dont just sit there and do nothing!

Erosion139, 2024-10-30:
### Cura Version

5.9.1 Beta

### Operating System

Win11

### Printer

Custom FFF

### Reproduction steps

Use scarf seam

### Actual results

before a scarf seam move the toolhead jumps up 0.2mm then back down 0.2mm for seemingly no reason causing an XY movement pause and apparent bump. It doesn't need to move Z, its already at where it needs to be. I tested this on the default 'draft' profile and only enabled a 15mm scarf. This behavior is on both the outer and inner walls. 

### Expected results

Scarf seams are supposed to mitigate travel pauses and consistent material over or undershoot when lifting to the next layer. But this is not mitigated if there are useless Z movements that especially dont move the X and Y. Print quality is reduced with scarf seams in this way. 

### Add your .zip and screenshots here ⬇️

[image]
[image]
[image]

https://github.com/user-attachments/assets/4309dd03-09fa-4bd0-928b-0a12f9597349

GregValiant, 2024-10-30:
Thanks for the report.
Scarf Seams remain a work in process.  I'll add this to the growing list of general _bugginess_ that it is causing.

See #19804 and #19814 which aren't quite duplicates but point to issues with Scarf Seam.
I'll mark this as a duplicate and go ahead and close it since the team is aware of the problems with the current implementation of Scarf Seams in 5.9.0beta.  Things like this are why there are beta's.

GregValiant, 2024-11-08:
@HellAholic - I still see this "up down" behavior in 5.9.0 Beta 2 in the travel moves.

;MESH:NONMESH
G0 F600 X59.211 Y112.668 **Z1.2** >>>>> move up to the height of the next layer
G0 F10500 X59.211 Y112.668 **_Z1_**  >>>>>> This value appears to be incorrect
G0 X61.386 Y119.057
....... Combing Moves
;TIME_ELAPSED:252.614817
;LAYER:5
M117 6/367 | ET 4h08m
M118 6/367 | ET 4h08m
G0 X142.704 Y127.2 **Z1.2** >>>>>>>>> correct layer height

[Scarf seam Z.zip](https://github.com/user-attachments/files/17677804/Scarf.seam.Z.zip)

GregValiant, 2024-11-08:
Here I am again.  A day late and a dollar short.

closed 2024-11-08 by GregValiant