Ultimaker/Cura #19834: getting this error when trying to slice: Oops! We encountered an unexpected error during your slicing process. Rest assured, we've automatically received the crash logs for analysis, if you have not disabled data sharing in your preferences. To assist us further, consider sharing your project details on our issue tracker.

ErikSattler, 2024-10-30:
### Cura Version

5.7.0

### Operating System

windows 10

### Printer

Creality Ender-3/Enders-3 v2

### Reproduction steps

loaded Mistral-Ev1.3_LEO_N and sliced it

### Actual results

Got this erro Oops! We encountered an unexpected error during your slicing process. Rest assured, we've automatically received the crash logs for analysis, if you have not disabled data sharing in your preferences. To assist us further, consider sharing your project details on our issue tracker.

### Expected results

wanted  a sample gcode file to export to my printer

### Add your .zip and screenshots here ⬇️

[Mistral-E Filament Cooling Duct - 3255518.zip](https://github.com/user-attachments/files/17571868/Mistral-E.Filament.Cooling.Duct.-.3255518.zip)

GregValiant, 2024-10-30:
Thanks for the report.
I didn't have any trouble slicing the model for my Ender 3 Pro in 5.8.0 or in 5.7.0 with my settings.

We ask for project files because they contain your settings and your printer as well as the model.  We ask for log files because they can contain details of why the slice aborted.  Please provide those.

This is in 5.7.0 on my Win10 laptop.
[image]

ErikSattler, 2024-10-30:
Where do I find the log files?

ErikSattler, 2024-10-30:
Hi GragValiant

I forgot to mention that I also rotated the model because I had issues with the strength of the mounting points. So my model was rotated 90 degrees so the output port was facing upwards. Where do I find the logs you need?

Best
Regards
Erik

image.png (view on web)<https://github.com/user-attachments/assets/3f6ef0f3-043e-4c5c-b78a-88211597e63f>

STATENS SERUM INSTITUT
5 Artillerivej | DK-2300 Copenhagen S | T +45 3268 3268 | F +45 3268 3868 | E ***@***.******@***.***> | W ssi.dk<https://www.ssi.dk>

This e-mail, inclusive of attachments, is intended for the person(s) or entity stated above and may contain confidential information. Unauthorised reading, disclosure, copying, distribution or use of this information may violate rights to proprietary information. If you are not an intended recipient, please return this e-mail to the sender and delete your copy. Thank you.

Read more about how SSI processes personal data (ENG)<https://en.ssi.dk/about-us/information-about-processing-of-personal-data>

Denne e-mail, inklusiv vedhæftede filer er tiltænkt den eller de personer, der er angivet ovenfor, og kan indeholde fortrolige oplysninger. Uautoriseret læsning, videregivelse, kopiering, distribution eller anden brug af disse oplysninger kan krænke rettighederne til beskyttede oplysninger. Hvis du ikke er en tiltænkt modtager, bedes du returnere denne e-mail til afsenderen og slette din kopi. Tak skal du have.

Læs mere om hvordan SSI behandler personoplysninger (DA)<https://www.ssi.dk/om-ssi/persondatapolitik>

GregValiant, 2024-10-30:
The instructions were on the form you filled out.
Set Cura up and use the "File | Save Project" command to create a project file.
Use the "Help | Show Configuration Folder" command to locate the Cura.log file.
Right after a slice fails - go to the log file and zip Cura.log.  Add the project file to the zip folder.  Post the zip here directly as email attachments won't come through.

GregValiant, 2024-10-30:
There are some fine features on that model.  Make sure that:
"Outer Wall Inset" = 0
"Hole Horizontal Expansion" = 0
"Horizontal Expansion" = 0
Also check the "Scaling" in the Material settings.

All of those can effect the slice and if they make walls too thin then you may get voids.

I'll go ahead and close this.

closed 2024-10-30 by GregValiant

ErikSattler, 2024-10-30:
Thanks a lot! I’ll try that.

Best Regards
Erik