Ultimaker/Cura #19837: [5.x] AppImage fails to start on debian trixie

t-cc, 2024-10-30:
### Cura Version

5.8.1

### Operating System

Linux debian trixie, kde desktop

### Printer

all printers

### Reproduction steps

Try to start cura from the App Image file on debian trixie

### Actual results

```
qt.qpa.gl: Xcb EGL gl-integration initialize failed
qt.glx: qglx_findConfig: Failed to finding matching FBConfig for QSurfaceFormat(version 4.1, options QFlags<QSurfaceFormat::FormatOption>(), depthBufferSize -1, redBufferSize 1, greenBufferSize 1, blueBufferSize 1, alphaBufferSize -1, stencilBufferSize -1, samples -1, swapBehavior QSurfaceFormat::SingleBuffer, swapInterval 1, colorSpace QColorSpace(), profile  QSurfaceFormat::CoreProfile)
qt.glx: qglx_findConfig: Failed to finding matching FBConfig for QSurfaceFormat(version 4.1, options QFlags<QSurfaceFormat::FormatOption>(), depthBufferSize -1, redBufferSize 1, greenBufferSize 1, blueBufferSize 1, alphaBufferSize -1, stencilBufferSize -1, samples -1, swapBehavior QSurfaceFormat::SingleBuffer, swapInterval 1, colorSpace QColorSpace(), profile  QSurfaceFormat::CoreProfile)
Could not initialize GLX
Abortado (`core' generado)
```

### Expected results

Cura starts

### Add your .zip and screenshots here ⬇️

[stderr.log](https://github.com/user-attachments/files/17576995/stderr.log)

ohagandavid, 2024-10-31:
I have this issue on a trixie install too. 5.6.0 and 5.8.1 appImages affected. Apt logs have a bunch of updates to qt6 on 10/26. libqt6opengl6?

witenite, 2024-11-02:
I am having the same issue on Trixie, with 5.8.1 and 5.9.0-Beta1

As a normal user and under Sudo....

susno762, 2024-11-03:
Same issue here with Debian Trixie and 5.8.1. Output of --debug attached.

[debug.log](https://github.com/user-attachments/files/17610932/debug.log)

TheRealOne78, 2024-11-05:
This seems to happen only on AMD GPUs, not necessarily just on trixie.

This happens even on the older `modern` builds

JeanLucCoulon, 2024-11-06:
I've an Intel CPU (core I7 6700K) with an integrated GPU (HD Graphics 530),
and I've the problem with debian sid.
It used to work (Ultimaker Cura 5.8.0) on the same hardware
Regards
Jean-Luc

madebyTimo, 2024-11-07:
I also have the problem on Intel GPU (8. gen CPU with Intel UHD Graphics 620).
On Nvidia GPU (RTX 3070) with proprietary drivers it works fine.

Both systems have latest Debian Trixie. The problem startet with an recent apt upgrade where some libqt6 packages were updated.

SpyroTEQ, 2024-11-09:
I just made an `apt dist-upgrade` on my Kali machine (so Debian based too), and got a Python crash on starting cura 5.0.0-3 (the software stats, but crashes right away):
```
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/UM/Qt/ListModel.py", line 54, in data
    return self._items[index.row()][self._role_names[role].decode("utf-8")]
           ~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
KeyError: 'material_name'
```

So I turned to latest AppImage, and got the same issue as described here (Could not initialize GLX) on a core2duo with a HD6770:
```
┌──(xenos㉿homefixe)-[~/Téléchargements]
└─$ md5sum ./UltiMaker-Cura-5.8.1-linux-X64.AppImage
16bd0c4a7afcb60f08b0da882095809d  ./UltiMaker-Cura-5.8.1-linux-X64.AppImage
                                                                                                                                                                                                                                              
┌──(xenos㉿homefixe)-[~/Téléchargements]
└─$ ./UltiMaker-Cura-5.8.1-linux-X64.AppImage       
Gtk-Message: 21:46:24.039: Failed to load module "canberra-gtk-module"
Gtk-Message: 21:46:24.044: Failed to load module "canberra-gtk-module"
qt.glx: qglx_findConfig: Failed to finding matching FBConfig for QSurfaceFormat(version 4.1, options QFlags<QSurfaceFormat::FormatOption>(), depthBufferSize -1, redBufferSize 1, greenBufferSize 1, blueBufferSize 1, alphaBufferSize -1, stencilBufferSize -1, samples -1, swapBehavior QSurfaceFormat::SingleBuffer, swapInterval 1, colorSpace QColorSpace(), profile  QSurfaceFormat::CoreProfile)
qt.glx: qglx_findConfig: Failed to finding matching FBConfig for QSurfaceFormat(version 4.1, options QFlags<QSurfaceFormat::FormatOption>(), depthBufferSize -1, redBufferSize 1, greenBufferSize 1, blueBufferSize 1, alphaBufferSize -1, stencilBufferSize -1, samples -1, swapBehavior QSurfaceFormat::SingleBuffer, swapInterval 1, colorSpace QColorSpace(), profile  QSurfaceFormat::CoreProfile)
Could not initialize GLX
zsh: IOT instruction  ./UltiMaker-Cura-5.8.1-linux-X64.AppImage
                                                                                                                                                                                                                                              
┌──(xenos㉿homefixe)-[~/Téléchargements]
└─$ cat /proc/cpuinfo 
processor	: 0
vendor_id	: GenuineIntel
cpu family	: 6
model		: 15
model name	: Intel(R) Core(TM)2 CPU          6600  @ 2.40GHz
stepping	: 6
microcode	: 0xd0
cpu MHz		: 2404.075
cache size	: 4096 KB
physical id	: 0
siblings	: 2
core id		: 0
cpu cores	: 2
apicid		: 0
initial apicid	: 0
fpu		: yes
fpu_exception	: yes
cpuid level	: 10
wp		: yes
flags		: fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ht tm pbe syscall nx lm constant_tsc arch_perfmon pebs bts rep_good nopl cpuid aperfmperf pni dtes64 monitor ds_cpl vmx est tm2 ssse3 cx16 xtpr pdcm lahf_lm pti tpr_shadow dtherm
vmx flags	: tsc_offset vtpr
bugs		: cpu_meltdown spectre_v1 spectre_v2 spec_store_bypass l1tf mds swapgs itlb_multihit mmio_unknown
bogomips	: 4808.15
clflush size	: 64
cache_alignment	: 64
address sizes	: 36 bits physical, 48 bits virtual
power management:

processor	: 1
vendor_id	: GenuineIntel
cpu family	: 6
model		: 15
model name	: Intel(R) Core(TM)2 CPU          6600  @ 2.40GHz
stepping	: 6
microcode	: 0xd0
cpu MHz		: 2404.071
cache size	: 4096 KB
physical id	: 0
siblings	: 2
core id		: 1
cpu cores	: 2
apicid		: 1
initial apicid	: 1
fpu		: yes
[... 66 lines cut ...]
[MainThread] UM.PackageManager.__init__ [56]: Found bundled packages JSON file: /tmp/.mount_UltiMaVHV0x7/share/cura/resources/bundled_packages/cura.json
[MainThread] UM.View.GL.OpenGLContext.detectBestOpenGLVersion [124]: Trying OpenGL context 4.1...
qt.glx: qglx_findConfig: Failed to finding matching FBConfig for QSurfaceFormat(version 4.1, options QFlags<QSurfaceFormat::FormatOption>(), depthBufferSize -1, redBufferSize 1, greenBufferSize 1, blueBufferSize 1, alphaBufferSize -1, stencilBufferSize -1, samples -1, swapBehavior QSurfaceFormat::SingleBuffer, swapInterval 1, colorSpace QColorSpace(), profile  QSurfaceFormat::CoreProfile)
qt.glx: qglx_findConfig: Failed to finding matching FBConfig for QSurfaceFormat(version 4.1, options QFlags<QSurfaceFormat::FormatOption>(), depthBufferSize -1, redBufferSize 1, greenBufferSize 1, blueBufferSize 1, alphaBufferSize -1, stencilBufferSize -1, samples -1, swapBehavior QSurfaceFormat::SingleBuffer, swapInterval 1, colorSpace QColorSpace(), profile  QSurfaceFormat::CoreProfile)
Could not initialize GLX
Fatal Python error: Aborted

Current thread 0x00007efe75526740 (most recent call first):
  File "UM/View/GL/OpenGLContext.py", line 36 in setContext
  File "UM/View/GL/OpenGLContext.py", line 125 in detectBestOpenGLVersion
  File "UM/Qt/QtApplication.py", line 178 in initialize
  File "cura/CuraApplication.py", line 348 in initialize
  File "UM/Application.py", line 288 in run
  File "UM/Qt/QtApplication.py", line 385 in run
  File "cura/CuraApplication.py", line 880 in run
  File "cura_app.py", line 243 in <module>

Extension modules: PyQt6.QtCore, PyQt6.QtNetwork, PyQt6.QtGui, PyQt6.QtWidgets, numpy.core._multiarray_umath, numpy.core._multiarray_tests, numpy.linalg.lapack_lite, numpy.linalg._umath_linalg, numpy.fft._pocketfft_internal, numpy.random._common, numpy.random.bit_generator, numpy. [... cut ...] geometry_helpers, scipy.ndimage._nd_image, _ni_label, scipy.ndimage._ni_label (total: 87)
zsh: IOT instruction  ./UltiMaker-Cura-5.8.1-linux-X64.AppImage --debug
```
Same with `./UltiMaker-Cura-5.9.0-beta.2-linux-X64.AppImage` and weirdly, with `Ultimaker-Cura-5.0.0-linux.AppImage` too (to 5.0.0-3 from apt boots then crashes, but the AppImage in the releases crashes before any screen shows up).

I kept trying older versions, and the `Ultimaker_Cura-4.13.0.AppImage` is working.

I hope that helps

witenite, 2024-11-10:
> I just made an `apt dist-upgrade` on my Kali machine (so Debian based too), and got a Python crash on starting cura 5.0.0-3 (the software stats, but crashes right away):
>
> ```
> Traceback (most recent [...]

My debug results looks pretty much the same.

boorcecom, 2024-11-12:
Same here on Debian trixie on a Lenovo Laptop.
Ryzen 7 5800H (amggpu) + Nvidia RTX4060, using wayland.
1st, need to run with QT_QPA_PLATFORM=xcb (to avoid an error with wayland...)

But it fails with the same error.
`QT_QPA_PLATFORM=xcb ./UltiMaker-Cura-5.8.1-linux-X64.AppImage --debug
trimesh/curvature.py:12: DeprecationWarning: Please use `coo_matrix` from the `scipy.sparse` namespace, the `scipy.sparse.coo` namespace is deprecated.
  from scipy.sparse.coo import coo_matrix
Gtk-Message: 11:34:35.056: Failed to load module "canberra-gtk-module"
Gtk-Message: 11:34:35.056: Failed to load module "canberra-gtk-module"
[MainThread] UM.Application.initialize [163]: Initializing UltiMaker Cura
[MainThread] UM.Application.initialize [164]: App Version 5.8.1
[MainThread] UM.Application.initialize [165]: Api Version 8.8.0
[MainThread] UM.Application.initialize [166]: Build type None
[MainThread] UM.PluginRegistry.addPluginLocation [205]: Plugin location /tmp/.mount_UltiMaANLMru/../Resources/share/uranium/plugins must be a folder.
[MainThread] UM.PluginRegistry.addPluginLocation [205]: Plugin location /tmp/.mount_UltiMaANLMru/../Resources/share/cura/plugins must be a folder.
[MainThread] UM.PluginRegistry.addPluginLocation [205]: Plugin location /tmp/lib/uranium must be a folder.
[MainThread] UM.PluginRegistry.addPluginLocation [205]: Plugin location /tmp/lib64/uranium must be a folder.
[MainThread] UM.PluginRegistry.addPluginLocation [205]: Plugin location /tmp/lib32/uranium must be a folder.
[MainThread] UM.PluginRegistry.addPluginLocation [205]: Plugin location /tmp/.mount_UltiMaANLMru/plugins must be a folder.
[MainThread] UM.PluginRegistry.addPluginLocation [205]: Plugin location /tmp/Resources/uranium/plugins must be a folder.
[MainThread] UM.PluginRegistry.addPluginLocation [205]: Plugin location /tmp/Resources/cura/plugins must be a folder.
[MainThread] UM.Resources.__initializeStoragePaths [481]: Initializing storage paths
[MainThread] UM.Resources.__initializeStoragePaths [491]: Config storage path is /home/ncazier@ad.boorce.com/.config/cura/5.8
[MainThread] UM.Resources.__initializeStoragePaths [499]: Data storage path is /home/ncazier@ad.boorce.com/.local/share/cura/5.8
[MainThread] UM.Resources.__initializeStoragePaths [511]: Cache storage path is /home/ncazier@ad.boorce.com/.cache/cura/5.8
[MainThread] UM.Qt.QtApplication.initialize [164]: Preferences file not found, ignore and use default language 'en_US'
[MainThread] UM.PackageManager.__init__ [56]: Found bundled packages JSON file: /tmp/.mount_UltiMaANLMru/share/uranium/resources/bundled_packages/uranium.json
[MainThread] UM.PackageManager.__init__ [56]: Found bundled packages JSON file: /tmp/.mount_UltiMaANLMru/share/cura/resources/bundled_packages/bundled_CuraEngineGradualFlow.json
[MainThread] UM.PackageManager.__init__ [56]: Found bundled packages JSON file: /tmp/.mount_UltiMaANLMru/share/cura/resources/bundled_packages/cura.json
[MainThread] UM.View.GL.OpenGLContext.detectBestOpenGLVersion [124]: Trying OpenGL context 4.1...
qt.glx: qglx_findConfig: Failed to finding matching FBConfig for QSurfaceFormat(version 4.1, options QFlags<QSurfaceFormat::FormatOption>(), depthBufferSize -1, redBufferSize 1, greenBufferSize 1, blueBufferSize 1, alphaBufferSize -1, stencilBufferSize -1, samples -1, swapBehavior QSurfaceFormat::SingleBuffer, swapInterval 1, colorSpace QColorSpace(), profile  QSurfaceFormat::CoreProfile)
qt.glx: qglx_findConfig: Failed to finding matching FBConfig for QSurfaceFormat(version 4.1, options QFlags<QSurfaceFormat::FormatOption>(), depthBufferSize -1, redBufferSize 1, greenBufferSize 1, blueBufferSize 1, alphaBufferSize -1, stencilBufferSize -1, samples -1, swapBehavior QSurfaceFormat::SingleBuffer, swapInterval 1, colorSpace QColorSpace(), profile  QSurfaceFormat::CoreProfile)
Could not initialize GLX
Fatal Python error: Aborted

Current thread 0x00007f74d82c2740 (most recent call first):
  File "UM/View/GL/OpenGLContext.py", line 36 in setContext
  File "UM/View/GL/OpenGLContext.py", line 125 in detectBestOpenGLVersion
  File "UM/Qt/QtApplication.py", line 178 in initialize
  File "cura/CuraApplication.py", line 348 in initialize
  File "UM/Application.py", line 288 in run
  File "UM/Qt/QtApplication.py", line 385 in run
  File "cura/CuraApplication.py", line 880 in run
  File "cura_app.py", line 243 in <module>

Extension modules: PyQt6.QtCore, PyQt6.QtNetwork, PyQt6.QtGui, PyQt6.QtWidgets, numpy.core._multiarray_umath, numpy.core._multiarray_tests, numpy.linalg.lapack_lite, numpy.linalg._umath_linalg, numpy.fft._pocketfft_internal, numpy.random._common, numpy.random.bit_generator, numpy.random._bounded_integers, numpy.random._mt19937, numpy.random.mtrand, numpy.random._philox, numpy.random._pcg64, numpy.random._sfc64, numpy.random._generator, scipy._lib._ccallback_c, scipy.sparse._sparsetools, scipy.sparse._csparsetools, scipy.sparse.linalg._isolve._iterative, scipy.linalg._fblas, scipy.linalg._flapack, scipy.linalg._cythonized_array_utils, scipy.linalg._flinalg, scipy.linalg._solve_toeplitz, scipy.linalg._matfuncs_sqrtm_triu, scipy.linalg.cython_lapack, scipy.linalg.cython_blas, scipy.linalg._matfuncs_expm, scipy.linalg._decomp_update, scipy.sparse.linalg._dsolve._superlu, scipy.sparse.linalg._eigen.arpack._arpack, scipy.sparse.csgraph._tools, scipy.sparse.csgraph._shortest_path, scipy.sparse.csgraph._traversal, scipy.sparse.csgraph._min_spanning_tree, scipy.sparse.csgraph._flow, scipy.sparse.csgraph._matching, sci [... cut ...] ython.src._highs_wrapper, scipy.optimize._highs._highs_wrapper, scipy.optimize._highs.cython.src._highs_constants, scipy.optimize._highs._highs_constants, scipy.linalg._interpolative, scipy.optimize._bglu_dense, scipy.optimize._lsap, scipy.optimize._direct, shapely.lib, shapely._geos, shapely._geometry_helpers, scipy.ndimage._nd_image, _ni_label, scipy.ndimage._ni_label (total: 87)
[1]    73878 IOT instruction (core dumped)  QT_QPA_PLATFORM=xcb ./UltiMaker-Cura-5.8.1-linux-X64.AppImage --debug
`

gigem, 2024-11-19:
This problem seams to be related to the libz3-4 package.  Cura works with version 4.8.12-3.1+b2 but not with version 4.8.12-3.1+b3 and later.  Because b3 is simply a rebuild with different tools and libraries, the real problem probably lies elsewhere but I've yet to find it.  My work around for now is to extract libz3.so.4 from the b2 package and put it in a custom LD_LIBRARY_PATH when running cura.

wklaebe, 2024-12-01:
Unpacking the AppImage, dropping the libz3.so.4 from libz3-4_4.8.12-3.1_amd64.deb into there, and starting Cura with ./AppRun works too.

rephlex, 2024-12-01:
> Where is the bug though? In libz3-4 or somewhere else?

Nah, it's just Cura being incompatible with a newer libz3-4 version, probably because of some function call change.

julienrobin28, 2024-12-06:
Hello there,

Just as a feedback, I can confirm those points (on Debian 13/Trixie up to date as 2024-12-06)

 - When using `amdgpu` driver, be it on Radeon 760M IGP or R9 280X GPU, `UltiMaker-Cura-5.9.0-linux-X64.AppImage` give the described error about "Failed to finding matching FBConfig for QSurfaceFormat"

 - When disabling the 760M IGP and forcing the NVIDIA GPU, using proprietary NVIDIA driver, no more error: it works.

 - Back on amdgpu, having downloaded the previous libz3-4 version `libz3-4_4.8.12-3.1_amd64.deb` package for Debian 12/Bookworm (and having it extracted manually into `/some/folder/old-libz3-4/`) Cura works fine when launched with `LD_LIBRARY_PATH=/some/folder/old-libz3-4/ ./UltiMaker-Cura-5.9.0-linux-X64.AppImage`

Congrats to @gigem for having found that!

vostok01, 2024-12-17:
I have no libz3 library in my gentoo distro at all, but this error is present. libz3 is missing both in my system and in UltiMaker-Cura-5.9.0-linux-X64.AppImage. Moreover, libz3 has never been installed at all.
Therefore, I don't really understand how Cura can depend on libz3.

ls-2020, 2024-12-18:
> I have no libz3 library in my gentoo distro at all, but this error is present. libz3 is missing both in my system and in [...]

same error here and no libz3! my guess is that this bug is related to **amdgpu** and not to libz3

rephlex, 2024-12-18:
> my guess is that this bug is related to **amdgpu** and not to libz3

I confirm this, as from multiple systems running the same package versions but various intel/nvidia/amd GPUs, only systems which use AMD GPUs are affected.

vostok01, 2024-12-18:
> only systems which use AMD GPUs are affected.

I also have an AMD GPU.

JeanLucCoulon, 2024-12-18:
I've a core i7-6700K with the integrated graphic attachment (Intel HD
Graphics 530/
And I've the problem.

Jean-Luc

TheRealOne78, 2024-12-18:
Do you also have an AMD dedicated GPU, or just the integrated GPU?

> I've a core i7-6700K with the integrated graphic attachment (Intel HD
> Graphics 530/
> And I've the problem.

JeanLucCoulon, 2024-12-18:
Nope

ls-2020, 2024-12-19:
@t-cc
please change the issue to:  **[5.x] AppImage fails to start on LINUX**

If **MESA** is compiled with Rusticl the AppImage fails to start, so many distributions are affected!

You can  downgrade (no nice solution) mesa to an older version and the AppImage will start again.

To fix this issue the code that initializes GLX should be updated to work with latest release of mesa.

ask6483, 2024-12-26:
I can confirm the problem on Debian trixie (testing) with Intel HD Graphics 620.
I also tried the fix mentioned here with older version of libz3 from package `libz3-4_4.8.12-3.1_amd64.deb`, this doesn't work.

ask6483, 2024-12-27:
I managed to build Cura from source and run it successfully. 

How to create appimage? Documentation is lacking support on this topic.

I have tried to use `crate_appimage.py` but I don't know what to pass via `dist_path` parameter.

kostnik, 2025-03-08:
same here on gentoo, cura 5.9.0 (https://github.com/BlueManCZ/edgets/blob/master/media-gfx/cura/cura-5.9.0.ebuild):

```sh
$ cura
Gtk-Message: 19:12:30.444: Failed to load module "xapp-gtk3-module"
Gtk-Message: 19:12:30.444: Failed to load module "canberra-gtk-module"
Fontconfig warning: "/etc/fonts/conf.avail/05-reset-dirs-sample.conf", line 6: unknown element "reset-dirs"
qt.glx: qglx_findConfig: Failed to finding matching FBConfig for QSurfaceFormat(version 4.1, options QFlags<QSurfaceFormat::FormatOption>(), depthBufferSize -1, redBufferSize 1, greenBufferSize 1, blueBufferSize 1, alphaBufferSize -1, stencilBufferSize -1, samples -1, swapBehavior QSurfaceFormat::SingleBuffer, swapInterval 1, colorSpace QColorSpace(), profile  QSurfaceFormat::CoreProfile)
qt.glx: qglx_findConfig: Failed to finding matching FBConfig for QSurfaceFormat(version 4.1, options QFlags<QSurfaceFormat::FormatOption>(), depthBufferSize -1, redBufferSize 1, greenBufferSize 1, blueBufferSize 1, alphaBufferSize -1, stencilBufferSize -1, samples -1, swapBehavior QSurfaceFormat::SingleBuffer, swapInterval 1, colorSpace QColorSpace(), profile  QSurfaceFormat::CoreProfile)
Could not initialize GLX
Aborted
```

theli-ua, 2025-04-07:
Just can't make it work here on gentoo on amd now at all. nor downgrading z3 nor mesa helped. Not sure how to workaround :(

fmgrotepass, 2025-06-03:
Crap! This problem returned for me. I'm running on Gentoo with an Intel onboard graphics card. Last time I solved it by downgrading  Mesa, but now things broke again. The image fix shared by @wjskrydl  did not work.

fmgrotepass, 2025-06-13:
I solved the problem for my machine. I did change 2 things at the same time: 

The one that I think solved it was disabling the llvm use flag for mesa. I also enabled the z3 flag (as it was also mentoined here) for llvm. Maybe if I have time I'll see which one causes the bug, but for now it works and the long queue of print jobs that need printing can finally be done! Gentoo Linux users have a solution how to tackle this and if newer versions of Ubuntu or other Linux distro's have this issue, then at least there is some information for a solution.

Liriel, 2025-06-17:
Can confirm: compiling mesa without `llvm` does the trick for me as well. Tested with `UltiMaker-Cura-5.9.0-linux-X64.AppImage` on Gentoo with `mesa-25.0.7`and `llvm-19.1.7`.

m-sundman, 2025-07-22:
This works for me:
`LD_LIBRARY_PATH=/folder/containing/libz3.so.4/file ./UltiMaker-Cura-5.10.1-linux-X64.AppImage`

mlepage, 2025-08-03:
Works for me too. To summarize (for those who might have trouble implementing the workaround):

- Download bookworm deb from one of the sites listed at https://packages.debian.org/bookworm/amd64/libz3-4/download
- Verify the contents (you'll need the path in the deb) with a command like: `dpkg --contents libz3-4_4.8.12-3.1_amd64.deb`
- Extract the deb with a command like: `dpkg-deb -x libz3-4_4.8.12-3.1_amd64.deb libz3-4_4.8.12-3.1_amd64`
- Run the AppImage with a command like `LD_LIBRARY_PATH=~/Downloads/usr/lib/x86_64-linux-gnu ./UltiMaker-Cura-5.10.2-linux-X64.AppImage` (assuming the deb was extracted in `~/Downloads`)

_This is only a workaround until the AppImage is fixed, of course._

cdp1337, 2025-08-17:
+1 to this issue too; just upgraded to Trixie and went to print something I was working on, only to realize the AppImage didn't load with that GLX error.

Didn't try the LD_LIBRARY_PATH work-around, but `sudo apt install cura` installed v5.0.0 from the repos, which seems to work well enough, (providing you don't need the latest features).

theli-ua, 2025-09-07:
Still can't make this to work on Gentoo, used to work if I recompile mesa without llvm, now it won't even let me do it as it'll  disable radeonsi driver

jimy-byerley, 2025-10-04:
same problem here on debian 13, @theli-ua 's built did the job

jeremywang19740412, 2025-11-10:
> This problem seams to be related to the libz3-4 package. Cura works with version 4.8.12-3.1+b2 but not with version 4.8.12-3.1+b3 and later. Because b3 is simply a rebuild with different tools and libraries, the real problem probably lies elsewhere but I've yet to find it.

Man, you are amazing, how did you find out, me can not figure this out myself until I see your solution (me even looked at the coredump file but no clue), thank you really~

HakanKoseoglu, 2026-01-02:
< tap tap> Is this project on? How on earth no one is picking this up and fixing it?

rephlex, 2026-01-02:
> > < tap tap> Is this project on? How on earth no one is picking this up and fixing it?
> 

It's probably one of
- WoRkS oN ArChLiNuX, wOnTfIx!!11
- nObOdY iS uSiNg DeBiAn AnYwAyS
- nObOdY iS uSiNg LuNiX AnYwAyS

rephlex, 2026-01-02:
same as #18891 #19837 #19913 #20353 #20706 #20806 #21089 #20516

rephlex, 2026-10-25:
On Debian trixie you can skip the AppImage entirely. My company runs a small signed APT repo where we build Cura natively against the distro's own libraries, because on Debian and Ubuntu the package manager already does everything an AppImage tries to do, only better: dependencies, unattended updates, menu entries, icons, file associations.

```
curl -fsSL https://mirror.plustech.de/plustech.gpg | gpg --dearmor -o /usr/share/keyrings/plustech.gpg
echo "deb [signed-by=/usr/share/keyrings/plustech.gpg] https://mirror.plustech.de/debian/ plustech-trixie main" > /etc/apt/sources.list.d/plustech.list
apt-get update && apt-get install ultimaker-cura
```

That gives you Cura 5.13.0 as a .deb built in a trixie chroot. No chmod +x, no FUSE, no AppImage runtime in the middle of your OpenGL stack. Sources are published on the site (LGPL/AGPL). Community repo, not an official UltiMaker channel.