Ultimaker/Cura #19840: Problem managing anything

MadScientistWI, 2024-10-31:
### Cura Version

5.9.0-beta.1

### Operating System

MAC 13.6.9 (22G830)

### Printer

Ender 3 S1 PRO

### Reproduction steps

I can open any management windows once only. Then I need to close Cura and reopen to get in the management again.

Next I am adding new materials, but it will not link to a Ender 3 profile, but it does to my AnyCubic Kobra MAX. It said the the profile is not supported!

### Actual results

Close Cura and open it again.

Material add is blank profile and not supported

### Expected results

It to work!

### Add your .zip and screenshots here ⬇️

Cura Problem.zip

GregValiant, 2024-11-01:
Thanks for the report.
You zip file did not attach.  The text is there but there is no hyperlink.  Give the file a chance to upload.  You cannot post files here when responding to an email notification.  You must post files directly to the site.

MadScientistWI, 2024-11-01:
here is the file of the screen shot again.
[Cura Problem.zip](https://github.com/user-attachments/files/17603554/Cura.Problem.zip)
Cura Problem.zip

GregValiant, 2024-11-02:
The "Not Supported" message often comes up when there is no "quality file" for a particular nozzle and material combination.  With the line width at 0.4 I would assume you have a 0.4 nozzle installed.  Try changing to "Generic PLA" and see if it will allow a slice.
Changes to quality are stored in the Configuration Folder ("Help | Show Configuration Folder").  I know they aren't easy to read but the file names may give you a clue.
Since you are running on a MAC I can't really comment on the "open once and then have to restart" issue.  That may indeed be a bug.
Have you tried re-installing Cura?
I think there is going to be a second beta release of 5.9.0 because of problems with the Scarf Seam code.  In the meantime are you having the same sort of issues with 5.8.0?

GregValiant, 2026-09-08:
I'll close this.

closed 2026-09-08 by GregValiant