Ultimaker/Cura #19842: z seam issue

LogaNN9, 2024-11-01:
### Cura Version

5.9.0

### Operating System

Windows 10

### Printer

Creality Cr6 Max

### Reproduction steps

Open Cura slicer with the recent update.
Enable the "Scarf Seam" feature.
Print a model, ensuring to observe the layer transitions.

### Actual results

After the first layer, each time the nozzle moves to a new layer, it pauses for around 10 seconds at the seam location while continuing to extrude filament before eventually moving.
Even when not using the "Scarf Seam" feature, there is still a pause whenever the nozzle transitions to a new layer, causing extra filament to be deposited at the seam location. This issue has existed even before the latest update. The nozzle does not begin moving immediately upon switching layers but instead briefly pauses, leading to filament buildup in that area.

### Expected results

The nozzle should move immediately after transitioning to a new layer without unnecessary pauses, ensuring no extra filament is deposited at the seam location, resulting in a cleaner print.

For your reference, I am using a Creality CR-6 Max printer. And I do not experience this issue with other slicer programs

Please address these problems, as they disrupt the printing process and severely affect print quality.

Thank you in advance for your assistance.

### Add your .zip and screenshots here ⬇️

[SHUTTLE CLOUD.zip](https://github.com/user-attachments/files/17599133/SHUTTLE.CLOUD.zip)
[image]
You can see the location where the nozzle is waiting in the attached photo

GregValiant, 2024-11-01:
Thanks for the report.
I generated a gcode from your project and I don't see any problem in it.

This is the Cura preview in "Speed" mode.  The speeds are consistent down low and then the "Minimum Layer Time" kicks in for the two peaks.  At that point the printer slows down.  There are no "pauses" or G4 lines in the gcode.
[image]

How are you sending the gcode to the printer?

LogaNN9, 2024-11-01:
I'm using an SD card to send the G-code to the printer. Also, I don't have
an issue with speed; as you can see from the image I sent, I noticed this
problem in the first few layers of the print

1 Kas 2024 Cum 18:46 tarihinde GregValiant ***@***.***> şunu
yazdı:

GregValiant, 2024-11-01:
I can't get the problem to occur.  I "air printed" (no extrusions) the gcode file and it was fine on my Ender 3 Pro.

Post the actual gcode file you had trouble with (zip and post it directly here - do not try to post it via email as it won't show up).  I'll take a look at it.  Maybe something is different.

I'm thinking the problem is likely with the printer, but I don't know how that could happen either.  It's unlikely that this is a Cura problem because you are the only person who has reported this sort of behavior.  Sorry I can't be more help.

LogaNN9, 2024-11-01:
I think that when I use this feature, the G-code file becomes too complex
for the printer, and it has trouble processing it. I plan to update the
printer’s firmware and try again. Thank you for your attention.

1 Kas 2024 Cum 22:39 tarihinde GregValiant ***@***.***> şunu
yazdı:

GregValiant, 2024-11-02:
_"...the G-code file becomes too complex..."_
Unless the mainboard hardware is really pathetic - I don't think that would be it.  Gcode is just instructions for the printer to move, or to adjust it's internal settings ("M" commands). 
The printer should accept them line-by-line, do the calculations from mm to steps, and turn the motors the correct distance.  The mainboard CPU's are pretty much the same across all of the inexpensive printers.

There are certain delays that can be caused by the gcode but they aren't really time delays.  If the "Maximum Resolution" setting in Cura is set really low (like at .01 for example) then a bazillion lines of gcode might be required for a 1mm extrusion.  That's going to take a while to complete and because the extrusions are so short the printer seems to be pausing because it just isn't moving very far.  Your "Maximum Resolution" setting of 0.4 should be fine so I doubt if that is the problem.

That isn't to say that you don't have a mainboard problem.  It could be anything from the card reader, to a chip problem.  Going from one version of Marlin to another isn't likely to make a difference.
if you make up a short gcode file simply consisting of the lines below, and print it as you normally would from the SD card, then it should reset the printer to the factory defaults

**M502** ; Reset to factory defaults
**G4 S2** ; Pause for 2 seconds so the printer processor has time to write to the SD.
**M500** ; Save the settings (to the SD on some printers or to the EEPROM on other printers).
**M300** ; Beep

I would do that and run a test print before playing with the firmware.  After resetting - if you had any customizations to the firmware (changes to things like the Max acceleration, the Home Offsets, Max feed rates, PID, etc.) they would need to be inputted again.

LogaNN9, 2024-11-02:
I ran the G-code you provided and tried again. The result is still the
same. The layer height of the model I printed was 0.2, which is standard
quality, so I don't think the resolution setting is causing this issue.
However, later on, when I tried printing an approximately 3 cm airplane
engine fan blade at a high resolution (0.08) with the scarf seam feature
active, I noticed something. Instead of staying completely still, the
nozzle was moving in very tiny steps at the seam location. This might be
happening because, as you mentioned, the G-code file contains too many
lines. However, I don’t understand why this issue occurred in my first
print, where I used a 0.6 nozzle with a 0.2 layer height.

2 Kas 2024 Cmt 05:32 tarihinde GregValiant ***@***.***> şunu
yazdı:

GregValiant, 2024-11-03:
There are definite bugs with the "Scarf Zseam".  They are being addressed by the Cura team and a fix should be in place in 5.9.0beta2.
In the meantime I would advise against using Scarf Zseam as it can't be trusted.  You can turn it off or slice with 5.8.0 or 5.8.1.

LogaNN9, 2024-11-03:
Thank you for taking my concerns into consideration. I appreciate your
prompt response and the information about the upcoming fix. I’ll avoid
using the Scarf Zseam feature for now and look forward to the improvements
in the next release.

3 Kas 2024 Paz 03:46 tarihinde GregValiant ***@***.***> şunu
yazdı:

LogaNN9, 2024-11-11:
Hello,

I’m still experiencing the "Scarf Seam" issue where the nozzle pauses
briefly at the seam on each layer, causing filament buildup and affecting
print quality. I have updated to the latest version, but the problem
persists.

For your information.

3 Kas 2024 Paz 13:17 tarihinde Atahan Günay ***@***.***> şunu
yazdı: