Ultimaker/Cura #19843: The wiping movement goes in the wrong direction

DerKob, 2024-11-01:
### Cura Version

5.8.1

### Operating System

Windows 10

### Printer

Artillery SWX1

### Reproduction steps

This error has occurred since many Cura versions. If you print any part with two or more walls, the wiping movement goes in the right direction. If you reduce the walls to only one, the wiping movement goes in the wrong direction and makes the Z-seam bigger. You can see the bug in the images I have uploaded. To show the error more clearly, I have increased the wiping distance to 1mm. Normally I use 0.2mm. 

Here are the instructions to reproduce the bug step by step. 

1. increase the wiping distance to 1mm to see the bug more clearly. 

2. do a test with two walls (the wiping process is correct).

3. reduce to one wall only. 

[image]
[image]

### Actual results

Now the wipe goes to the wrong directory and makes the Z seam larger instead of smaller. Once the wall has been completed, the nozzle moves backwards. 

### Expected results

When the wall is finished, the nozzle should travel a little beyond the starting point to hide the Z-seam. 

### Add your .zip and screenshots here ⬇️

[image]
[image]
[image]
[image]
[ASX1_Distanzstueck Savage STL.zip](https://github.com/user-attachments/files/17599174/ASX1_Distanzstueck.Savage.STL.zip)

The uploaded file is different from the part in the uploaded screenshots. The bug is the same.

GregValiant, 2024-11-01:
Thanks for the report.
I can reproduce this with my own models.
The Cura team has been working on the Zseam.  Maybe @HellAholic has some information on this behavior?

DerKob, 2024-11-05:
Thank you for your help. Can you estimate in how many weeks/months the official version 5.9.0 will be released?

DerKob, 2024-11-12:
Thank you for fixing it.