Ultimaker/Cura #19851: Printing to the front left corner and off of bed on x,y axis

KyEz96, 2024-11-02:
### Cura Version

5.81

### Operating System

Windows 11 home

### Printer

Ender 5 Plus

### Reproduction steps

Slice any file and when you print it goes to the front left corner to start print instead of in the center. 

### Actual results

Prints begins not in the center. I have made sure the "origin at center" box is not checked as suggested. In the program the piece is centered and when using creality slicer it prints correctly in the center of the bed. I would like to use cura though but I cant figure out what is wrong.

### Expected results

Print should begin in center of bed

### Add your .zip and screenshots here ⬇️

Happens with all files

GregValiant, 2024-11-03:
Thanks for the report.
Without a project file it's impossible to tell what is going on.  There are thousands of Ender 5's out there and this is the first report concerning this.

It is most likely that this is a problem with your printer settings or your Cura settings.
Open the Gcode file in a text editor.  At the beginning will be the area that is used on the build plate.  MINX and MINY should both be positive numbers.  If they are negative, then the "Origin at Center" box is not checked, or it is a bug in the setting box code which no one else has.

The printer will put the "print origin" at the Home Offset 0,0,0 so if your printer "Home Offsets" are incorrect then the print can be off the build surface.  Right after "auto-homing" the printer, the LCD should display the current location as it relates to the Home Offset origin.  Those are in fact your Home Offset numbers.

"Origin at Center" should NOT be checked in Cura as your printer origin is (I believe) at the right rear corner when you stand in front of the printer.  (If you stand behind the printer then you will see that the origin is indeed the left front corner.  The Ender 5's and 6's are different because of the way the hardware is laid out.)
I suggest you install another instance of the Ender 5 in Cura and see if it makes a difference.

I made this platform for a friend for his Ender 5.  He hasn't had any trouble with prints being off the build surface.
[image]

KyEz96, 2024-11-03:
Thank you very much for replying! I hope we can get this sorted out. I actually started googling the issue first and it seems there are quite a few people with the same exact issue but with enders in general not just the 5 plus like mine. They seem to not resolve it in the forums though or some of them have the box checked which I do not.
It prints in the center of the bed when using creality slicer and also on Orca, but in Cura it goes the the front left. 
Yes my 5 Plus starts in the back right corner but when opening the ender 5 plus profile in cura it shows the front left unlike the picture you attached the shows the back right? I will attach a file and also some screenshots to this email. I tried to include the file in the initial report but it said it was too big.
You said to try to install another instance of the ender 5... how do I do that? I only see 1 "Ender 5 Plus"  that comes up when I go to "add printer" It wont let me send the g code it says it is too big

 
 

image.png (view on web)

KyEz96, 2024-11-03:
I'll try to send it in acompressed file

GregValiant, 2024-11-03:
Just add another Ender 5.  You can have multiple instances of the same printer model loaded.  Cura will assign a name like "Ender 5 #2" so you can tell which is which.
In File Explorer, right click on the 3mf project file and select "Send To... / Compressed (zipped) folder".  Post the zip folder here. 
 Any model will work.  A calibration cube won't be a large file.  You can zip a gcode file as well.

I've been doing triage here for 4 1/2 years and every time this has come up it's been a user issue.  The default build plate "origin" in Cura is Left Front Corner.  There is nothing in any Creality definition files that change that.

closed 2024-12-10 by github-actions[bot]