Ultimaker/Cura #19852: Z axis Gcode 0

MurderouslyCold, 2024-11-03:
### Cura Version

5.9.0-beta-1

### Operating System

windows 10

### Printer

creality ender 3 pro

### Reproduction steps

i wanted to slice bridging test, and noticed (while printing rantom movement in z axis (nozzle hited print multiple times) so i checked gcode and saw that in some lines before going up (0,5 finished should go to 0,7) there is Z0.00 and in next line there is z0,7.  i have tried on 5.8 and it did not happened.  there is file with test prints, and cura profile.

[cura issue.zip](https://github.com/user-attachments/files/17610578/cura.issue.zip)

### Actual results

there should be no collision

### Expected results

nozzle not going down to z0.00

### Add your .zip and screenshots here ⬇️

[image]

GregValiant, 2024-11-03:
Thanks for the report.
That looks like one of the problems that "Scarf Seam" introduced.  In the "Experimental" section, turn off the Scarf Seam by setting the "Scarf Seam Length" to 0.  You should be good.  The problem does not exist in the 5.8.x versions so you can revert to one of those as well.
I'll mark this as a duplicate and go ahead and close it.  The fix for this (and the other known Scarf Seam bugs) is in place and there should be a 5.9.0 Beta 2 release coming out with the fixes in it.

MurderouslyCold, 2024-11-03:
mine scarf seam length is/was set to 0.0 but if it is reported i will close this

GregValiant, 2024-11-03:
The profile you included has the Z seam length at 4.0.  That's why I mentioned it.
Slicing with 5.8.x is fine.  There were other things changed for 5.9.0 but Scarf Seam is a problem.
It's important for people to check the beta releases so things like this get caught.

MurderouslyCold, 2024-11-03:
i now checked that parameter again, i have set as 0.0 so there can be also visual bug or sth?

GregValiant, 2024-11-03:
Turning off Scarf Seams (setting the Scarf Length to 0.0) removes all problems associated with Scarf Seams.