Ultimaker/Cura #19854: Unterstützung für TEVO Michelangelo 3D-Drucker in Cura hinzufügen

JT401, 2024-11-03:
### Is your feature request related to a problem?

Hallo zusammen,
Ich möchte meinen TEVO Michelangelo 3D-Drucker in der neuesten Version von Cura integrieren. Leider ist der vorhandene .ini-Datei des Herstellers veraltet und wird von der aktuellen Cura-Version nicht erkannt.

### Describe the solution you'd like

Problemstellung:
•	Fehlendes Profil: Der TEVO Michelangelo ist nicht in der Liste der unterstützten Druckerprofile in Cura enthalten.
•	Veraltete .ini Datei: Ich besitze die originale .ini-Datei vom Hersteller, jedoch ist diese Datei zu alt und kompatibel mit den neuesten Cura-Versionen. Cura erkennt den Drucker dadurch nicht korrekt.
•	Technische Unterstützung: Mir fehlt das technische Wissen, um die .ini-Datei selbstständig zu aktualisieren oder ein neues Druckerprofil zu erstellen.
Bereitgestellte Dateien:
•	Original .ini Datei 

Vielen Dank im Voraus für eure Unterstützung! Ich hoffe, dass es dem Projekt hilft, den TEVO Michelangelo in Cura zu integrieren, sodass mehr Nutzer von diesem Drucker profitieren können.

### Describe alternatives you've considered

Keine

### Affected users and/or printers

TEVO Michelangelo 3D-Drucker

### Additional information & file uploads

[Michelangelo_ini.zip](https://github.com/user-attachments/files/17611563/Michelangelo_ini.zip)

GregValiant, 2024-11-04:
The printer definitions must come from an owner (such as yourself) or from the printer manufacturer.  UltiMaker cannot put resources into figuring out what works and what does not for another manufacturers products.  (It would be necessary to buy one of every printer made on the planet.)
There is a [GitHub page](https://github.com/Homers3D/Tevo-Michelangelo) with some links for support and such.  Checking with TEVO to see if they have a project file or Cura definition files would give you a big start into setting the printer up in UM Cura.

The function of any printer definition is to set Cura up so it knows things like the Firmware Flavor, the physical size of the build surface, where the "Origin" is on the printer, whether it has a heated bed, etc.  All of those ( and a lot more) are settings in Cura that can also be adjusted by the user.
That INI file was produced by Slic3R slicer in 2018 so as you found out, it won't work with Cura.

What you can do is to install a generic "CustomFFF" printer and then go through the Machine Settings and make changes for your Michaelangelo.  If you assume that the firmware is "Marlin" you will have about a 90% chance of being correct.

GregValiant, 2024-11-05:
I tried to find some information on this printer but it appears that the TEVO Michaelangelo has been out of production for a few years.  I'll go ahead and close this as it won't be addressed by UltiMaker.  I hope you understand.

closed 2024-11-05 by GregValiant