Ultimaker/Cura #19859: Coasting on 5.9.0 Beta causes a Z move to 0.00 on every wall.

tripartist1, 2024-11-05:
### Cura Version

5.9.0 Beta

### Operating System

Windows 10

### Printer

Ender 3v2 Klipper

### Reproduction steps

1. Load default profile
2. Enable coasting
3. Slice the model
4. Open gcode in editor, ctrl-f "z0.0"
5. Find a number of hits approximately equal to layer count

### Actual results

Coasting creates G0 moves to Z=0 on every coast instead of correctly coasting. If done on later layers (a few mm even) it will cause mechanical stress on the machine from attempting to travel THROUGH the printed part.

The built in gcode preview does NOT show these moves. You must use a gcode viewer, open the gcode file manually, or print the file to see the problem.

### Expected results

Coasting works as intended.

### Add your .zip and screenshots here ⬇️

None required, simply enable coasting.

GregValiant, 2024-11-05:
Thanks for the report.
This is due to a bug in "Scarf Seam".  Set the "Scarf Team Length" to 0.0 and the problem should go away.

I'll mark this as a duplicate and go ahead and close it.  The Cura team is aware of the problems associated with Scarf Seam and the fix is in place for 5.9.0 beta 2.

closed 2024-11-05 by GregValiant

tripartist1, 2024-11-07:
> Thanks for the report. This is due to a bug in "Scarf Seam". Set the "Scarf Team Length" to 0.0 and the problem should go away.
> 
> I'll mark this as a duplicate and go ahead and close it. The Cura team is aware of the problems associated with Scarf Seam and the fix is in place for 5.9.0 beta 2.

I have confirmed the problem does **not** go away with seam length set to 0.0. As it stands I am unable to make the problem go away without restarting the software. I have coasting, wiping, and scarf all completely disabled and the gcode still contains moves to z0.00 on every outer wall. It seems once the issue presents itself it does not like to go away. 

It may still pertain to scarf seams and the changes made to implement them, but simply disabling the feature doesn't help here.

GregValiant, 2024-11-08:
This was a problem in beta 1.  Did you try with beta 2?