Ultimaker/Cura #19860: Scarf Seams + Outer Wall Wipe Distance not working as expected

tripartist1, 2024-11-05:
### Cura Version

5.9.0 Beta

### Operating System

Windows 10

### Printer

Ender 3v2 Klipper

### Reproduction steps

1. Load a default draft profile
2. Enable "Outer wall wipe distance" by setting its value to >0 (I set to 5mm for visibility)
3. Enable Scarf Seams by setting "Scarf Seam Length" to >0 (I set to 10mm for visibility)
4. Slice any model (I used a basic cylinder from "parts for calibration")

### Actual results

Gcode is generated to 

1. Complete the scarf seam
2. Move back to the start of the wall (in my case 10mm) 
3. Complete the outer wall wipe move

### Expected results

Same as actual, minus #2:
1. Complete the scarf seam
2. DO NOT move back to the start of the wall (in my case 10mm) 
3. Complete the outer wall wipe move

### Add your .zip and screenshots here ⬇️

None needed, simple changes from default profiles.

GregValiant, 2024-11-05:
Thanks for the report.
Scarf Seam is being worked on for the beta II release.

@HellAholic - this may be something to interest you.

MariMakes, 2024-11-19:
👋 Quick update on our side.
This is resolved in the 5.9 release 🎉 , you can download the Cura version with the fix here: [https://github.com/Ultimaker/Cura/releases/tag/5.9.0](https://github.com/Ultimaker/Cura/releases/tag/5.9.0)

I'll close this issue since it's resolved.
Thanks again, and please let us know if you run into any other issues 💪

closed 2024-11-19 by MariMakes