Ultimaker/Cura #19862: Bug no pós processamento

ramon202424, 2024-11-05:
### Cura Version

5,8,1

### Operating System

Windows 11 

### Printer

Ender 3v2 neo

### Reproduction steps

Pegue um projeto e coloque para fazer troca de cor no pós processamento 

### Actual results

Chegando na camada desejada da troca de cor o comando e feito levando a cabeca de imprenssao para o canto da mesa logo apos chegar no local seguro da troca do filamengo o eixo Z aparece como 0 no monitor ... acionando para voltar a imprenssao o mesmo nao reconhece mais a altura do eixo Z assim entao comecando a imprimir no ar por que conta do 0 que estava na para para cima e nao volta na imprenssao que esta na mesa 

### Expected results

deveria acontecer da cabeca de imprenssao voltar a imprimir na imprensao que ja estava na mesa .obs usei o prusa slicer e a imprensora reconheceu o eixo Z na para de troca de cor normalmelte ... espero que resolvam isso gosto muito de usar o cura ...

### Add your .zip and screenshots here ⬇️

.

GregValiant, 2024-11-06:
Thanks for the report.
We ask for project files so we can check all the settings and see what happened.  I use PauseAtHeight a lot and I am familiar with Filament Change.  Filament Change passes all the parameters to the firmware which then decides how to handle the pause.  PauseAtHeight does not depend on the firmware.  It would be good to know which one you are having trouble with.  No one else has reported any problems and neither script has been changed for several versions.

Please post a project file.  You cannot post files by email so you will need to zip the project and post it directly here.  You might post the problem Gcode as well.  More information is better.

ramon202424, 2024-11-06:
Testei a opção Filamento Change aconteceu o mesmo erro começa a imprimir quando manda o comando para troca de cor ela subiu até os 20mm depois quando voltei a impressão novamente voltou a imprimir no ar com menos distância do que a outra opção mais mesmo assim não volta na impressão

ramon202424, 2024-11-06:
Não tem como ter nada de errado com meus arquivos são arquivos que já imprimi outra vezes mais de 10 vezes e sempre dava certo e também não acredito que seja algo com sensor do equipamento já que ele faz o zeramento normal

GregValiant, 2024-11-06:
I cannot duplicate the problem using my settings.  I don't see any problem in the code for PauseAtHeight or FilamentChange (which at any rate work completely differently) and yet you have the problem with both of them.
I'll close this now because without a project file nothing further can be done.  If you provide one then this can be re-opened.  I hope you understand.

closed 2024-11-06 by GregValiant