Ultimaker/Cura #19866: Automatically switch to extruder 2 when extruder 1’s material is depleted in Ultimaker S5 printer

EHM2018, 2024-11-06:
### Is your feature request related to a problem?

In Ultimaker S5 printer, material in extruder 1 is depleted and I am not at that moment present to change it.

### Describe the solution you'd like

To have an option in Cura to automatically switch to extruder 2 when extruder 1’s material is depleted, allowing the print job to continue seamlessly.

### Describe alternatives you've considered

To buy an Ultimaker s5 Pro Bundle Material Station Auto Material Reload

### Affected users and/or printers

Ultimaker S5 printer

GregValiant, 2024-11-07:
@EHM2018 I'm going to go ahead and close this.  It isn't a matter or "won't do" but rather "can't do".  Once a gcode file is generated it is "carved in stone" as it were.  There is simply no way to interject anything into it.

If the printer notices that the filament runs out - it can of course pause the gcode stream while it does something else, and then it can resume from where it left off (provided the XYZE are re-synced to the resume location in the gcode).  That all has to take place within the printer firmware.  The gcode has no input nor flexibility for something like that.  It just isn't possible.

closed 2024-11-07 by GregValiant

GregValiant, 2024-11-08:
_"What I don't know about UM printers fills libraries"_, but on the surface (and to a dummy) this does seem like a good idea as a firmware option.  (I don't think you would want it enabled by default, or for every print.  You wouldn't want a print that ended up half PVA).

@fieldOfView - Is there a UM site or forum to make this suggestion?