Ultimaker/Cura #19869: No profiles/printers/setting after upgrading from 5.8.0 to 5.9.0-beta2

Marooned-MB, 2024-11-06:
### Cura Version

5.9.0-beta2

### Operating System

Windows 10 Home

### Printer

Artillery Sidewinder X3 Pro

### Reproduction steps

1. I've downloaded and installed Cura 5.9.0-beta2
2. After running, Cura shows welcome screen like it was never run before and ask me to add my 1st printer
3. I have also installed Cura 5.8.0 with 2 printers and a few custom profiles
4. I was unable to make 5.9.0 to transfer that data [I've tried copying user\AppData\Roaming\cura\5.8\ to 5.9 folder, nothing changed]
5. I'm logged in UltiMaker account in both version

### Actual results

No data in 5.9.0-beta2

### Expected results

All printers, profiles and settings from existing 5.8.0

### Add your .zip and screenshots here ⬇️

-

GregValiant, 2024-11-07:
Thanks for the report.
I always cross my fingers and close my eyes when I upgrade Cura.  Occasionally there are issues.  I upgraded from beta1 to beta2 an hour ago and everything came through fine.  This is on a Win10Pro system.  I've never had an Ultimaker account.
@Asterchades may have a better take on install problems.

Asterchades, 2024-11-07:
My first instinct in this case would be to see if it's repeatable. For two reasons: one to rule out a momentary blip causing a problem, and two to get a fresh log file which hopefully points in the right direction for the cause. I'd do this by removing the 5.9 directory entirely and letting it behave as if it actually was just a fresh installation, seeing what (if anything) it imports automatically from the existing 5.8 installation and how it behaves.

Only launch it once after having done that. If it still has an issue, upload the log (%appdata%\cura\5.9\cura.log) here for a more thorough inspection. I can't say for sure I'll be able to figure it out from that but there's going to be people here who can, or can at least glean enough to figure out an appropriate next step.

I was unable to reproduce this myself after multiple attempts. However that doesn't necessarily mean a huge amount as we're going to have different systems with different configurations - straight up I use Win11 and don't log my Cura in, so it's possible one or both of those could be an influencing factor (or have nothing at all to do with it).

Marooned-MB, 2024-11-07:
I'm trying to attach the log, but after adding a file I can't see it anywhere. No error, no file.

Inside log file can see it sees previous data and copy it:
```
2024-11-07 18:48:19,561 - DEBUG - [MainThread] UM.Resources._copyLatestDirsIfPresent [540]: Latest config path: C:\Users\Marooned\AppData\Roaming\cura\5.8 and latest data path: C:\Users\Marooned\AppData\Roaming\cura\5.8
2024-11-07 18:48:19,561 - INFO - [MainThread] UM.Resources.copyVersionFolder [576]: Copying directory from 'C:\Users\Marooned\AppData\Roaming\cura\5.8' to 'C:\Users\Marooned\AppData\Roaming\cura\5.9'
```

However, later in that file I can see an error regarding my new printer (I have too):
```
2024-11-07 18:53:32,678 - ERROR - [MainThread] UM.Settings.ContainerStack.deserialize [453]: When trying to deserialize Artillery Sidewinder X3 Pro, we received an unknown container ID (artillery_sidewinder_x3_pro)
```
Not sure if that error is the reason I have not data in 5.9.0? And how to fix that?
Everything in 5.8.0 works fine.

Asterchades, 2024-11-07:
Okay, I think I know what's happened here - or at least I can replicate the same error, which gives me reason to suspect I've figured out the cause.

There's no built-in definition for the *Artillery Sidewinder X3 Pro*, so my suspicion is that you've added a definition for it from another source. Trouble is that you've added it directly into your Cura installation directory (default _C:\Program Files\Ultimaker Cura 5.8.0_ for 5.8.0). Files within the Cura program directory don't migrate between versions, meaning your migrated config is looking for a definition that doesn't exist in the new installation. Since the definition doesn't exist it cannot be "serialised" (or anything else, for that matter), and the fail-safe kicks in asking you to reset your configuration.

Note that there should have been a background (shadowed) message similar to this visible if your screen resolution is high enough (I run the Dark theme, and my test definition was simply _Test.def.json_ and called _Test Printer_):
[image]

You have two choices to fix it:

First would be to copy the files from your 5.8.0 directory (or the originally downloaded definition files) into your 5.9.0 beta directory. You'll want everything which starts with _artillery\_sidewinder\_x3_pro_ from every directory under _share\cura\resources_, being sure to maintain the directory structure. This is not a good idea, as if this is the cause then the same problem will simply keep recurring every time you try to update your local install - including going to the 5.9.0 release.

Second, and this is preferable, is to use the config directory (eg _%appdata%\cura\5.8_) to store your custom definitions. The definition files all go into the same subdirectory as they would have under _share\cura\resources_ directory within the Cura installation (so _share\cura\resources\definitions_ to _%appdata%\cura\5.8\definitions_ and so forth). Like the rest of your config files these do get migrated between versions.

Marooned-MB, 2024-11-07:
Yay! That was it! Thanks
Yes, I've added non existing printer following some webpage. I've moved mentioned files from /definitions and /variants from Program Files to User directory 5.8 as you suggested and it was successfully moved to 5.9.

I did not have that error message you pasted - it went straight to welcome screen like there was no data to transfer from the previous installation at all.

closed 2024-11-07 by Marooned-MB

Asterchades, 2024-11-07:
Weird that the error didn't appear (perhaps I have something configured to show it) but at least it's working now.

Happy printing!