Ultimaker/Cura #19870: Bug Troca De Cor

ramon202424, 2024-11-07:
### Cura Version

5,8,1

### Operating System

Windows 11 

### Printer

Ender 3v2 neo

### Reproduction steps

Qualquer Arquivo persiste no mesmo erra ja contatei o suporte aqui e me disseram que esta tudo normal com os gcode .
Coloco para imprimir e adiciono o pós processamento de troca de cor pause at height .

### Actual results

após as etapas de configuração ao chegar no ponto da troca a cabeça de impressão vai ate a areá designada para troca assim que chega no lugar ela perde a referencia do eixo Z e passa a entender que aquela altura e o ponto 0 do Z após das inicio para continuar a impressão ela volta imprimindo por cima da peça no ar ..... me disseram que sou louco e que não existe esse problema no fatiador cura ... utilizei o Prusa Slicer  e a troca de cor funcionou normal arrumem esse erro eu não sou demente e entendo do que estou falando ...

### Expected results

imprimir de volta na peça na mesa não no ar

### Add your .zip and screenshots here ⬇️

[erro.zip](https://github.com/user-attachments/files/17654757/erro.zip)

GregValiant, 2024-11-07:
Thanks for the report.
I don't see anything wrong in the gcode either.  This is part of the pause code...
```
;current layer: 5
M83 ; switch to relative E values for any needed retraction
G1 F300 Z2.56 ; move up a millimeter to get out of the way
G1 F9000 X190 Y190
G1 F300 Z15 ; too close to bed--move to at least 15mm
M104 S200 ; standby temperature
M84 S3600 ; Keep steppers engaged for 1h   <<<<<<<<<  stepper disarm time of 1 hour
M300 S440 P2000 ; Beep
M0 ; Do the actual pause
M109 S200 ; WAIT for resume temperature
G1 F300 Z1.56
G1 F9000 X126.524 Y115
G1 F300 Z1.56 ; move back down to resume height
G1 F976 ; restore extrusion feedrate
M82 ; switch back to absolute E values
G92 E60.37684
;LAYER:5
```
"M84 S3600" tells the printer to keep the stepper motors active for 1 hour (3600 seconds) after the last movement command is received.  If the machine is ignoring that then the steppers can "disarm" after the default time in the firmware which is often only 2 minutes.  At that point they lose their position.
If you happen to be away from the printer when the pause occurs, and you come back 2 hours after the pause, then the steppers will have disarmed during your absence because the "1 hour time out" expired.  The maximum "disarm time" on my Ender 3 Pro is 14400 seconds (4 hours).
If you select the "Keep motors engaged" checkbox then you get Cura's default of 1 hour for M84.
If you de-select "Keep motors engaged" and instead enter 14400 into the "Disarm timeout" textbox then the steppers should not lose position for 4 hours.

I don't see a bug here as the Gcode is exactly right if you had checked "Keep motors engaged" in Pause At Height.  I use Pause At Height a lot and I have never had an issue with the steppers disarming and losing position.

ramon202424, 2024-11-07:
Ah cara vocês tão fazendo hora com a minha cara né ? Você viu o arquivo que eu fiz ele não demora nem 10 minutos direito a ser feito como estou demorando a fazer a troca de cor depois de 4 horas? eu acompanhei o processo no programa que dura alguns minutos ... Você e um péssimo profissional vou abrir uma reclamação sobre você!

GregValiant, 2024-11-07:
If you want to fix it you may have to admit that it's possible that there is something going on with the printer or the firmware.  If you search around here you will see that this is the only bug report concerning this behavior in the last couple of years.  The problem is not in the Gcode and so must be in the manner that your printer is responding to it.  PauseAtHeight has not been changed since 2022.  It happened to be a change that I suggested (and implemented) and that was accepted by UltiMaker.

You are right, I'm not a professional, I'm a volunteer (and not associated with Ultimaker) and so I'll volunteer this...if you go into this with blinders on and thinking that it MUST be a software issue then it's possible you will never find the problem.  That's why I explained all of the possibilities of what be occurring.  I will admit that I missed one and that is M18.

I'll go ahead and close this as you are being abusive and this isn't a Cura problem anyway.  Good luck.

closed 2024-11-07 by GregValiant