Ultimaker/Cura #19876: Retract before outer wall doesnt work

navi21, 2024-11-08:
### Cura Version

5.8.1

### Operating System

windows 10

### Printer

creality cr10

### Reproduction steps

check retract before outer wall, slice model then print

### Actual results

while printing its visible that it doesnt retract, it leaks while traveling from infill to outer wall, and outer wall has a gap because nozzle looses pressure before it needs to start outer wall

### Expected results

retraction before outer wall works and there is no artifacts on outer wall

### Add your .zip and screenshots here ⬇️

https://we.tl/t-gHL6ind3c6

GregValiant, 2024-11-08:
Thanks for the report.
I can duplicate this in both 5.8.0 and 5.9.0beta2.

@HellAholic might have time to take a look.

GregValiant, 2024-11-10:
In regards to things happening at the Zseam in 5.9.0-beta.2
With Z-hops turned off and Scarf Seam Length set to "0.0", I get this movement at the end of every Layer.
;MESH:NONMESH
G0 F600 X160.4 Y69.4 **Z66.5**
G0 F10500 X160.77 Y69.67 **Z66.25**
then
;LAYER:265
;MESH:Timer Cover.stl
G0 X161.6 Y69.9 **Z66.5**
That up-down-shuffle is unexpected and unwanted.

MariMakes, 2024-11-19:
👋 Quick update on our side.
This is resolved in the 5.9 release 🎉 , you can download the Cura version with the fix here: [https://github.com/Ultimaker/Cura/releases/tag/5.9.0](https://github.com/Ultimaker/Cura/releases/tag/5.9.0)

I'll close this issue since it's resolved.
Thanks again, and please let us know if you run into any other issues 💪