Ultimaker/Cura #19877: Easy way to block fuzzy skin for threaded shapes (holes)

RobWickham, 2024-11-08:
### Is your feature request related to a problem?

I have a simple test model with a single threaded hole for a screw to go into. But, fuzzy skin is happening to the threads in this hole, making the threads less effective.

### Describe the solution you'd like

I need some step by step for how to prevent this.

### Describe alternatives you've considered

I've looked at all the related issues and workarounds and can't make sense of them. I've temporarily placed a thin material cover over the entrance to the hole to force it to be "internal", and will drill into this to remove it after the print. That's a lousy workaround though.
[Test Print.zip](https://github.com/user-attachments/files/17681711/Test.Print.zip)

### Affected users and/or printers

Almost anyone using fuzzy skin with threaded holes will absolutely want this to be improved.

RobWickham, 2024-11-08:
[Partial Fuzzy Skin Test Print.zip](https://github.com/user-attachments/files/17682235/Partial.Fuzzy.Skin.Test.Print.zip)
Adding another test model using a thin cover over the threaded hole to block fuzzy skin -- but notice that the fuzzy skin STOPS being generated around the model where this "cover" exists and I suspect that's a bug. With a thicker "cover" (having more wall lines) doesn't have this behavior. Using a thicker cover is just that much more difficult to remove after the print completes.

GregValiant, 2024-11-09:
When the threads are exposed they are part of the "outside" and so they get fuzzy skin.
I know this is a test model but in this instance both "Enable Bridge Settings" and "Adaptive Layers" are confusing the issue.  Adaptive Layers in particular does not work well with Fuzzy Skin.

With those turned off, and a support blocker "Cutting Mesh" in place to get rid of the cap, I've put a second support blocker into the threaded hole and it's configured as a Cutting Mesh with the Fuzzy Skin setting turned off.  That results in this which sounds like what you want:
[image]

and the x-ray preview looks like this (I used a cylinder inside but a cube would do the same).
[image]

Fuzzy skin often requires some head-scratching to get right.

They aren't always possible but the preference is to use push-in threaded inserts, to make a pocket in the model and use Pause at Height to insert a nut, or to print the threads vertically.  Chasing the threads with a tap is often required to get rid of artifacts and the Zseam.

It's often a fine line between something being a bug, and it just being an old fashioned PITA.  I think this qualifies as a PITA rather than something broken in the software.