Ultimaker/Cura #19878: Model sliced fine in 5.8.0, fails in 5.9.0-beta2

Marooned-MB, 2024-11-08:
### Cura Version

5.9.0-beta2

### Operating System

Windows 10 Home

### Printer

Artillery Sidewinder X3 Pro

### Name abnormal settings

Default settings, speed bumped to 100mm/s

### Describe model location

Default after opening a file

### Describe your model

Sliced fine in 5.8.0

### Add your .zip here ⬇️

Public model: https://www.thingiverse.com/thing:4608059/files
file: among-keychain.stl

Here you can see comparison.
On the left: Cura 5.8.0, sliced fine
On the right: Cura 5.9.0-beta2, slicing error in aprox. 30% of progress bar with the error below:
```
Oops! We encountered an unexpected error during your slicing process. Rest assured, we've automatically received the crash logs for analysis, if you have not disabled data sharing in your preferences. To assist us further, consider sharing your project details on our issue tracker.
```
[image]

GregValiant, 2024-11-08:
Thanks for the report.
Without knowing your Cura settings it's hard to tell what might be happening.  I didn't have any trouble slicing this using my settings.
In 5.9.0 - use the "Help | Export package for technical support" command.  When it has finished, open the folder, right-click on the file and select "Send To... | Compressed (zipped) folder" and post the zip folder here.

Marooned-MB, 2024-11-08:
Ah, of course. Here is the exported package.
[export-20241108-212815.3mf.zip](https://github.com/user-attachments/files/17683780/export-20241108-212815.3mf.zip)

GregValiant, 2024-11-09:
Thank you.
Well, it still slices fine for me.  I'm using a Win10 laptop to slice with 5.9.0 beta 2.  I tried with 3 different printers and there was no problem.
Someone from the Cura team will take a look.

Marooned-MB, 2024-11-09:
So it sounds like it will affect anyone with a printer that is not default in Cura and I assume it could be a massive number? (I have a quite popular model)

Can it fallback to some default value for non existing parameters? If not, maybe some easy steps how to update manually such printer definitions put in upgrade instructions?

Marooned-MB, 2024-11-11:
Not sure where I got that definition from so it would be hard to contact the author.

I've looked into that definition file and compared it to for example `artillery_sidewinder_x1.def.json` that exist in Cura installation and they both looks quite similar:
* they inherit from `artillery_base`, also existing in Cura
* they have `metadata` and `overrides` section
So I would assume they both should work in similar way, so inheriting lacking params from `artillery_base`, which inherits from `fdmprinter`.

I´ve move fields around in X3 Pro json for cleaner diff and here is how it looks now:
[image]
I'm not familiar with the schema of those json files but doesn't look like there is some breaking differences. Any clue? Thanks

GregValiant, 2024-11-11:
The X3Pro definition within your project file is some sort of bad joke.  You can try your custom definition (which looks fine to me), or you can use the X1 definition and change the name and the rest of the settings within Cura without having to add nozzle files and quality files and maybe an extruder file.  None of that is terribly difficult but I think it can be avoided.  After you set up with the X1 you could just save a Project File and if your printer gets over-written somehow, just load that project file.

If there is a setting that is in error in the definition file, it would seem to me that any problem should have shown up when Cura loaded.  I don't think it should have resulted in an aborted slice.  I just seems that if there was a problem then why start the slice?