Ultimaker/Cura #19879: Wont slice

johnhack99, 2024-11-08:
### Cura Version

5.6.0

### Operating System

Windows 11

### Printer

S5

### Reproduction steps

added file to slice, didnt slice

### Actual results

didnt slice

### Expected results

sliced

### Add your .zip and screenshots here ⬇️

[image]

Asterchades, 2024-11-09:
5.6 is quite dated at this point. Every new version of Cura will address certain slicing issues as they've found, with 5.7 in particular having corrected an issue that was causing a large number of such failures (in addition to adding more advanced, optional, automatic reporting).

Since any fix involved would require updating anyway that's probably going to be the best place to start. 5.8.1 is the current version, with 5.9 in open beta.

GregValiant, 2024-11-09:
@johnhack99  The slicing problems of 5.6 (and 5.5) involved models with "organic" shapes like your octopus, and also models of buildings that had stone floors or roughhewn wood floors (lots of "islands" in one or two layers).  Those problems were addressed with 5.7.0.  As Asterchades mentioned - upgrading to a newer version like 5.8.x should fix the problem for you.

I'll mark this as a duplicate bug report and go ahead and close it.  I hope you understand.

closed 2024-11-09 by GregValiant