Ultimaker/Cura #19881: support blocker appears on the camera and not on the object

jimippu, 2024-11-09:
### Cura Version

5.8.1

### Operating System

windows 10

### Printer

ender 3 neo

### Reproduction steps

tried placing a support blocker on an object

### Actual results

support blocker appears directly on top of the viewing camera, when i move the camera, the blocker is far away from the object

### Expected results

for the support blocker to be on the object where i clicked it, happened in previous versions as well

### Add your .zip and screenshots here ⬇️

[image]

GregValiant, 2024-11-09:
Thanks for the report.
I have never seen this behavior in any of the 13 Cura versions I have installed.  This is on a Win10 systems as well.
The normal behavior is that the model must be "selected" and then the support blocker must be placed on the model.  Only then should the support blocker become part of the scene on the Cura build plate.
A workaround would be to select the support blocker (maybe from the Object List in the lower left corner) and then the "Translate Tool" ( the Move tool).  You can manually set the location to 0,0,0 to at least bring it onto the build plate.

When you bring a support blocker in, and then delete the model, does the support blocker disappear as well?  It is supposed to be part of the model and should be deleted along with the model.

Recreate the problem and then zip the cura.log file and post the zip folder here.  Maybe there will be a clue in it.

jimippu, 2024-11-10:
@GregValiant 
yes the support blocker gets deleted after i delete the model. i also found out that it only does the bug when its a model that i made myself in onshape as an stl file, also when i export the same self made part as a 3mf file, the support blockers work just normal.

GregValiant, 2024-11-10:
@jimippu , I'm not familiar with OnShape, but it seems there might be something amiss in the OnShape STL export utility.

@HellAholic you have led me to a new "Peter Maxx" effect.  Adding a blocker while the right-click menu was open gave me this.
[image]

I think it's because the camera is inside the block as I saw the polka-dot pattern for a split second before it evolved into that mess.
The block is correctly sized at 10x10x10, but it's in the Oort Cloud.
[image]