Ultimaker/Cura #19885: Slice faling with 5.8.1

berrabozz, 2024-11-09:
### Cura Version

5.8.1

### Operating System

MacOS 15.1

### Printer

Ultimaker 2+ Extended

### Describe your model

Very simple water tight model

### Add your .zip here ⬇️

[CasingLid.stl.zip](https://github.com/user-attachments/files/17689537/CasingLid.stl.zip)

GregValiant, 2024-11-10:
Thanks for the report.
We ask for project files and log files because without your settings we often can't duplicate the problem.
This is both right side up and upside down.  I could not make it "not slice".
[image]

berrabozz, 2024-11-10:
Hi there. I solved it. It turned out to be the NOD32 Antivirus that for some reason blocked some sockets used between internal components of Cura. NOD32 judged Cura as been infected by some virus. You should look into this, as it might cause issues for  others also. Found this WARNING in the log that led mw to the solution:

WARNING - [MainThread] UM.Backend.Backend._onSocketError [267]: Unhandled socket error ErrorCode.AcceptFailedError

GregValiant, 2024-11-10:
Good catch.
This was a problem with some previous versions of Cura but I seem to recall the installation was blocked rather than something happening further down the road (like you are getting).
I'll leave the bug label on this.  Someone from the Cura team will take a look.