Ultimaker/Cura #19886: Using a negative value for horizonal expansion causes infill to not generate in thin walls

5o6, 2024-11-10:
### Cura Version

5.8.1

### Operating System

Windows 10

### Printer

Ender 3 (Klipper, All-metal hotend, z probe)

### Reproduction steps

Slice a model with thin walls (less than 3mm) with a horizonal expansion value under 0 (-0.05 in my case). Notice missing infill.

### Actual results

Infill does not generate within the walls 

### Expected results

Infill should generate within the walls

### Add your .zip and screenshots here ⬇️

[image]
[CE3PRO_Bolt Pinning Jig - Imperial.zip](https://github.com/user-attachments/files/17690339/CE3PRO_Bolt.Pinning.Jig.-.Imperial.zip)

GregValiant, 2024-11-10:
Thanks for the report.
I don't think this really had anything to do with Horizontal Expansion other than it dropped the size of the "model feature" below a point that affected the infill generator filter.  This looks like a duplicate of #19478 (and #19802).
A repair has been made in 5.9.0-beta.2.
This is layer 455 in 5.8.1
[image]

And this is layer 455 in 5.9.0-beta.2  (the beta.1 release does not have this fix in it).
[image]

I'll mark this as a duplicate and go ahead and close it.  I hope you understand.

closed 2024-11-10 by GregValiant

5o6, 2024-11-10:
Thanks for the quick response, I'll go ahead and use the beta for now then.