Ultimaker/Cura #19891: slicing failed

georggggg, 2024-11-11:
### Cura Version

5.8

### Operating System

Windows 10

### Printer

Anycubic Chirron

### Reproduction steps

slicing failed - updated cura - slicing failed

### Actual results

slicing failed

### Expected results

sliced part

### Add your .zip and screenshots here ⬇️

[AC_Knopfzelle-Body_20241015.zip](https://github.com/user-attachments/files/17701119/AC_Knopfzelle-Body_20241015.zip)
[cura.log](https://github.com/user-attachments/files/17701150/cura.log)

georggggg, 2024-11-11:
I found out that the error is caused by my profile settings. I will use the standard profile.

georggggg, 2024-11-11:
fixed (comment bevor)

GregValiant, 2024-11-11:
Thanks for the report.
From your log file is the generic error "Backend exited abnormally with return code 3221225725!".  Whether supports are enabled or not, the model should slice.
Cura 5.8.0 would not slice your model.  From my log file is the same "Backend exited abnormally with return code 3221225725!"
I moved the model on the build plate and it sliced.
[image]

Someone from the Cura team will take a look.