Ultimaker/Cura #19894: Missing first layer

neonflux75, 2024-11-11:
### Cura Version

5.7.0, 5.8.1, 5.9.0 beta 2

### Operating System

Windows 11

### Printer

Artillery Genius

### Reproduction steps

Import in Cura the 20mm test cube.
The model autodrop in the plate.
Set the layer height to 0.2
Slice

### Actual results

Now we have a 99 layers sliced piece

### Expected results

I was expected 100 layers and, instead, i have only 99
I've done the test again with a piece that had a base of 0.4mm in heigh so i expected to found 2 layers as base and, instead, i had only one. The first layer was missing.
To have 2 layers i had to raise the model position in Z from 0 to 0.4 (with 0.2 the layer is still missing)
I've done the same things on the old 5.6.0 and all was fine

### Add your .zip and screenshots here ⬇️

[First layer missing bug.zip](https://github.com/user-attachments/files/17703872/First.layer.missing.bug.zip)

GregValiant, 2024-11-11:
Thanks for the report.
We ask for project files because your settings may have an effect on things like this.
You can see here that with the Initial Layer Height and the Layer Height set to 0.20 that I get 100 layers with your cube.
[image]

But if I set the Initial Layer Height to 0.30 I only get 99 layers.  The default "Initial Layer Height" for your Genius is indeed 0.30.
[image]

Even with both Initial Layer Height and Layer Height at 0.20, If I have Adaptive Layers enabled there will only be 83 layers.
[image]

I will remove the bug label for now as I have never had this problem and none of the other hundreds of thousands of Cura users has reported this with any version.  Please post a project file.  If it indicates a problem in Cura the bug label can be put back on.

neonflux75, 2024-11-11:
Ok, i've checked and you are right.
In the 5.6.0 the initial layer height is 0.2, in the other versions that I've tested (5.7.0, 5.8.1, 5.9.0 beta 2) this value is 0.3
Since I don't use this param I leave hidden by default so I've not see that was the reason of the "issue".
Sorry for the unuseful report and thanks for your time.
You can close or delete the post

closed 2024-11-11 by GregValiant