Ultimaker/Cura #19897: Cura 5.8.1 won't slice at all

Rayfincle, 2024-11-12:
### Cura Version

5.8.1

### Operating System

Windows 11

### Printer

Sovol SV07

### Name abnormal settings

Print will not slice on any device i try it on. Have tried other models and printers to no avail.

### Describe model location

No

### Describe your model

Yes, multiple times in Orca. Its a Cal cube.

### Add your .zip here ⬇️

[xyz-10mm-calibration-cube.zip](https://github.com/user-attachments/files/17721463/xyz-10mm-calibration-cube.zip)
[SS_xyz-10mm-calibration-cube - Shortcut.zip](https://github.com/user-attachments/files/17721502/SS_xyz-10mm-calibration-cube.-.Shortcut.zip)

GregValiant, 2024-11-12:
Thanks for the report.

Are we talking about Sovol Cura, or Ultimaker Cura?
Ultimaker Cura does not support the SV07 (yet).  If your problem is with Sovol Cura then they may have their own support site and the bug report would need to go there.

If you are using UM Cura:
Right after slicing fails - use the "Help | Show Configuration Folder" command to locate the "cura.log" file.  Zip that file and post the zip folder here.  Maybe there will be a clue as to what is happening on your system.
A project file would also be helpful as no one with UltiMaker Cura will have the SV07 printer definition files.

Rayfincle, 2024-11-13:
Hi Greg,

 

It is Ultimaker Cura I am using, and there are some config files on the Sovol website that can be used to add the SV07 to Cura and others are using it for their Sovols.  I have attached the log file here.

 

 

Even if I use the Ultimaker printer profile to slice it make no difference and so the issue appears to be agnostic of the printer.

 

 

 

 

Log Excerpt

 

2024-11-13 08:13:24,271 - INFO - [MainThread] UM.Qt.QtApplication.__init__ [89]: Adding QT6 plugin path: C:\Program Files\UltiMaker Cura 5.9.0-beta.2\PyQt6\plugins

2024-11-13 08:13:24,272 - INFO - [MainThread] UM.Qt.QtApplication.__init__ [99]: Adding QT5 plugin path: C:\Program Files\UltiMaker Cura 5.9.0-beta.2\PyQt6\plugins

2024-11-13 08:13:24,272 - DEBUG - [MainThread] UM.Application.initialize [163]: Initializing UltiMaker Cura

2024-11-13 08:13:24,273 - DEBUG - [MainThread] UM.Application.initialize [164]: App Version 5.9.0-beta.2

2024-11-13 08:13:24,273 - DEBUG - [MainThread] UM.Application.initialize [165]: Api Version 8.9.0

2024-11-13 08:13:24,274 - DEBUG - [MainThread] UM.Application.initialize [166]: Build type None

2024-11-13 08:13:24,274 - WARNING - [MainThread] UM.PluginRegistry.addPluginLocation [205]: Plugin location C:\Program Files\UltiMaker Cura 5.9.0-beta.2\..\Resources\share\uranium\plugins must be a folder.

2024-11-13 08:13:24,275 - WARNING - [MainThread] UM.PluginRegistry.addPluginLocation [205]: Plugin location C:\Program Files\UltiMaker Cura 5.9.0-beta.2\..\Resources\share\cura\plugins must be a folder.

2024-11-13 08:13:24,275 - WARNING - [MainThread] UM.PluginRegistry.addPluginLocation [205]: Plugin location C:\Program Files\lib\uranium must be a folder.

2024-11-13 08:13:24,276 - WARNING - [MainThread] UM.PluginRegistry.addPluginLocation [205]: Plugin location C:\Program Files\lib64\uranium must be a folder.

2024-11-13 08:13:24,276 - WARNING - [MainThread] UM.PluginRegistry.addPluginLocation [205]: Plugin location C:\Program Files\lib32\uranium must be a folder.

2024-11-13 08:13:24,276 - WARNING - [MainThread] UM.PluginRegistry.addPluginLocation [205]: Plugin location C:\Program Files\UltiMaker Cura 5.9.0-beta.2\plugins must be a folder.

2024-11-13 08:13:24,277 - WARNING - [MainThread] UM.PluginRegistry.addPluginLocation [205]: Plugin location C:\Program Files\Resources\uranium\plugins must be a folder.

2024-11-13 08:13:24,277 - WARNING - [MainThread] UM.PluginRegistry.addPluginLocation [205]: Plugin location C:\Program Files\Resources\cura\plugins must be a folder.

2024-11-13 08:13:24,278 - DEBUG - [MainThread] UM.Resources.__initializeStoragePaths [481]: Initializing storage paths

2024-11-13 08:13:24,278 - DEBUG - [MainThread] UM.Resources.__initializeStoragePaths [491]: Config storage path is C:\Users\KemicalAli\AppData\Roaming\cura\5.9

2024-11-13 08:13:24,278 - DEBUG - [MainThread] UM.Resources.__initializeStoragePaths [499]: Data storage path is C:\Users\KemicalAli\AppData\Roaming\cura\5.9

2024-11-13 08:13:24,279 - DEBUG - [MainThread] UM.Resources.__initializeStoragePaths [511]: Cache storage path is C:\Users\KemicalAli\AppData\Local\cura\5.9\cache

2024-11-13 08:13:24,280 - INFO - [MainThread] UM.PackageManager.__init__ [56]: Found bundled packages JSON file: C:\Program Files\UltiMaker Cura 5.9.0-beta.2\share\uranium\resources\bundled_packages\uranium.json

2024-11-13 08:13:24,280 - INFO - [MainThread] UM.PackageManager.__init__ [56]: Found bundled packages JSON file: C:\Program Files\UltiMaker Cura 5.9.0-beta.2\share\cura\resources\bundled_packages\cura.json

2024-11-13 08:13:24,281 - DEBUG - [MainThread] UM.View.GL.OpenGLContext.detectBestOpenGLVersion [124]: Trying OpenGL context 4.1...

2024-11-13 08:13:24,281 - DEBUG - [MainThread] UM.View.GL.OpenGLContext.detectBestOpenGLVersion [132]: Yay, we got at least OpenGL 4.1 core: 4.1 Core profile

2024-11-13 08:13:24,281 - DEBUG - [MainThread] UM.Qt.QtApplication.initialize [192]: Detected most suitable OpenGL context version: 4.1 Core profile

[... 298 lines cut ...]
2024-11-13 08:13:31,508 - WARNING - [MainThread] UM.Settings.ContainerRegistry.loadAllMetadata [497]: Invalid metadata for container deltabot_extruder_0: {'container_type': <class 'UM.Settings.DefinitionContainer.DefinitionContainer'>, 'id': 'deltabot_extruder_0', 'name': 'Extruder 1', 'version': 2, 'type': 'extruder', 'author': 'Ultimaker', 'manufacturer': 'Unknown', 'setting_version': 20, 'visible': False, 'position': '0', 'inherits': 'fdmextruder', 'machine': 'deltabot'}

2024-11-13 08:13:31,510 - WARNING - [MainThread] UM.Settings.ContainerRegistry.loadAllMetadata [497]: Invalid metadata for container geeetech_A30Pro: {'container_type': <class 'UM.Settings.DefinitionContainer.DefinitionContainer'>, 'id': 'geeetech_A30Pro', 'name': 'Geeetech A30Pro', 'version': 2, 'type': 'machine', 'author': 'Teddy.Hu', 'manufacturer': 'Geeetech', 'setting_version': 20, 'file_formats': 'text/x-gcode', 'visible': True, 'has_materials': True, 'has_variants': True, 'has_machine_quality': True, 'preferred_material': 'generic_pla', 'preferred_quality_type': 'standard', 'machine_extruder_trains': OrderedDict([('0', 'Geeetech_Single_Extruder')]), 'supports_usb_connection': True, 'supports_network_connection': False, 'inherits': 'Geeetech_Base_Single_Extruder', 'first_start_actions': ['MachineSettingsAction'], 'preferred_variant_name': '0.4mm Nozzle', 'quality_definition': 'Geeetech_Base_Single_Extruder', 'variants_name': 'Nozzle Size'}

[... cut ...]
2024-11-13 08:13:31,535 - WARNING - [MainThread] UM.Settings.ContainerRegistry.loadAllMetadata [497]: Invalid metadata for container farm2: {'container_type': <class 'UM.Settings.DefinitionContainer.DefinitionContainer'>, 'id': 'farm2', 'name': 'Farm 2', 'version': 2, 'type': 'machine', 'author': 'Farm3D', 'manufacturer': 'Farm3D LLC', 'setting_version': 20, 'file_formats': 'text/x-gcode', 'visible': True, 'has_materials': True, 'has_variants': True, 'has_machine_quality': False, 'preferred_material': 'generic_pla', 'preferred_qualit

LilBub, 2024-11-13:
One thing that caught my eye is you said you are using 5.8.1 but that large data dump shows 5.9.2 beta?

Rayfincle, 2024-11-13:
Yeah,

 

I have tried the beta version and 5.8.1 it’s the same across both versions. I can get the dump from 5.8.1 log 

 

2024-11-13 14:00:16,042 - INFO - [MainThread] UM.Qt.QtApplication.__init__ [89]: Adding QT6 plugin path: C:\Program Files\UltiMaker Cura 5.8.1\PyQt6\plugins

2024-11-13 14:00:16,043 - INFO - [MainThread] UM.Qt.QtApplication.__init__ [99]: Adding QT5 plugin path: C:\Program Files\UltiMaker Cura 5.8.1\PyQt6\plugins

2024-11-13 14:00:16,043 - DEBUG - [MainThread] UM.Application.initialize [163]: Initializing UltiMaker Cura

2024-11-13 14:00:16,044 - DEBUG - [MainThread] UM.Application.initialize [164]: App Version 5.8.1

2024-11-13 14:00:16,044 - DEBUG - [MainThread] UM.Application.initialize [165]: Api Version 8.8.0

2024-11-13 14:00:16,045 - DEBUG - [MainThread] UM.Application.initialize [166]: Build type None

2024-11-13 14:00:16,045 - WARNING - [MainThread] UM.PluginRegistry.addPluginLocation [205]: Plugin location C:\Program Files\UltiMaker Cura 5.8.1\..\Resources\share\uranium\plugins must be a folder.

2024-11-13 14:00:16,046 - WARNING - [MainThread] UM.PluginRegistry.addPluginLocation [205]: Plugin location C:\Program Files\UltiMaker Cura 5.8.1\..\Resources\share\cura\plugins must be a folder.

2024-11-13 14:00:16,046 - WARNING - [MainThread] UM.PluginRegistry.addPluginLocation [205]: Plugin location C:\Program Files\lib\uranium must be a folder.

2024-11-13 14:00:16,046 - WARNING - [MainThread] UM.PluginRegistry.addPluginLocation [205]: Plugin location C:\Program Files\lib64\uranium must be a folder.

2024-11-13 14:00:16,047 - WARNING - [MainThread] UM.PluginRegistry.addPluginLocation [205]: Plugin location C:\Program Files\lib32\uranium must be a folder.

2024-11-13 14:00:16,047 - WARNING - [MainThread] UM.PluginRegistry.addPluginLocation [205]: Plugin location C:\Program Files\UltiMaker Cura 5.8.1\plugins must be a folder.

2024-11-13 14:00:16,048 - WARNING - [MainThread] UM.PluginRegistry.addPluginLocation [205]: Plugin location C:\Program Files\Resources\uranium\plugins must be a folder.

2024-11-13 14:00:16,048 - WARNING - [MainThread] UM.PluginRegistry.addPluginLocation [205]: Plugin location C:\Program Files\Resources\cura\plugins must be a folder.

2024-11-13 14:00:16,049 - DEBUG - [MainThread] UM.Resources.__initializeStoragePaths [481]: Initializing storage paths

2024-11-13 14:00:16,049 - DEBUG - [MainThread] UM.Resources.__initializeStoragePaths [491]: Config storage path is C:\Users\KemicalAli\AppData\Roaming\cura\5.8

2024-11-13 14:00:16,049 - DEBUG - [MainThread] UM.Resources.__initializeStoragePaths [499]: Data storage path is C:\Users\KemicalAli\AppData\Roaming\cura\5.8

2024-11-13 14:00:16,050 - DEBUG - [MainThread] UM.Resources.__initializeStoragePaths [511]: Cache storage path is C:\Users\KemicalAli\AppData\Local\cura\5.8\cache

2024-11-13 14:00:16,050 - INFO - [MainThread] UM.PackageManager.__init__ [56]: Found bundled packages JSON file: C:\Program Files\UltiMaker Cura 5.8.1\share\uranium\resources\bundled_packages\uranium.json

2024-11-13 14:00:16,051 - INFO - [MainThread] UM.PackageManager.__init__ [56]: Found bundled packages JSON file: C:\Program Files\UltiMaker Cura 5.8.1\share\cura\resources\bundled_packages\bundled_CuraEngineGradualFlow.json

2024-11-13 14:00:16,051 - INFO - [MainThread] UM.PackageManager.__init__ [56]: Found bundled packages JSON file: C:\Program Files\UltiMaker Cura 5.8.1\share\cura\resources\bundled_packages\cura.json

2024-11-13 14:00:16,051 - DEBUG - [MainThread] UM.View.GL.OpenGLContext.detectBestOpenGLVersion [124]: Trying OpenGL context 4.1...

2024-11-13 14:00:16,052 - DEBUG - [MainThread] UM.View.GL.OpenGLContext.detectBestOpenGLVersion [132]: Yay, we got at least OpenGL 4.1 core: 4.1 Core profile

2024-11-13 14:00:16,052 - DEBUG - [MainThread] UM.Qt.QtApplication.initialize [192]: Detected most suitable OpenGL context version: 4.1 Core profile

2024-11-13 14:00:16,053 - INFO - [MainThread] UM.Qt.QtApplication.initialize [199]: Initializing job queue ...

2024-11-13 14:00:16,053 - INFO - [MainThread] UM.Qt.QtApplication.initialize [203]: Initializing version upgrade manager ...

2024-11-13 14:00:16,054 - INFO - [MainThread] UM.PackageManager._loadManagementData [174]: Loaded bundled packages data from C:\Program Files\UltiMaker Cura 5.8.1\share\uranium\resources\bundled_packages\uranium.json

2024-11-13 14:00:16,054 - INFO - [MainThread] UM.PackageManager._loadManagementData [174]: Loaded bundled packages data from C:\Program Files\UltiMaker Cura 5.8.1\share\cura\resources\bundled_packages\bundled_CuraEngineGradualFlow.json

2024-11-13 14:00:16,054 - INFO - [MainThread] UM.PackageManager._loadManagementData [174]: Loaded bundled packages data from C:\Program Files\UltiMaker Cura 5.8.1\share\cura\resources\bundled_packages\cura.json

2024-11-13 14:00:16,055 - INFO - [MainThread] UM.PackageManager._loadManagementData [201]: Loaded user packages management file from C:\Users\KemicalAli\AppData\Roaming\cura\5.8\packages.json

2024-11-13 14:00:16,055 - INFO - [MainThread] UM.PackageManager._removeAllScheduledPackages [280]: Attempting to remove the following scheduled packages: - 

[... 298 lines cut ...]
2024-11-13 14:00:24,042 - WARNING - [MainThread] UM.Settings.ContainerRegistry.loadAllMetadata [497]: Invalid metadata for container lotmaxx_sc10_extruder_0: {'container_type': <class 'UM.Settings.DefinitionContainer.DefinitionContainer'>, 'id': 'lotmaxx_sc10_extruder_0', 'name': 'Extruder 1', 'version': 2, 'type': 'e [... cut ...]  3 500mm', 'version': 2, 'type': 'machine', 'author': 'nu-hin', 'manufacturer': 'RatRig', 'setting_version': 20, 'file_formats': 'text/x-gcode', 'visible': True, 'has_materials': True, 'has_variants': True, 'has_machine_quality': True, 'preferred_material': 'generic_pla', 'preferred_quality_type': 'standard', 'machine_extruder_trains': OrderedDict([('0', 'ratrig_base_extruder_0')]), 'supports_usb_connection': True, 'supports_network_connection': False, 'inherits': 'ratrig_vcore3_base', 'exclude_

GregValiant, 2024-11-13:
A lot of those errors look like they have something to do with the installation.
I loaded the SV07 definition from the Sovol site and everything worked fine.
I was wondering if you have the plugin "StartUp Optimizer" running?

Try is run Cura "As Administrator" and see if it helps.
You might have to do a "clean install" by un-installing Cura, then manually deleting the Configuration Folder, and then re-installing Cura.