Ultimaker/Cura #19901: Propellor error

CaptainBrick, 2024-11-13:
### Cura Version

5.8.0

### Operating System

Windows 10

### Printer

Ultimaker 3

### Reproduction steps

1. load a model of a propellor for my Lego ship
2. https://grabcad.com/library/3-blade-propeller-5
3. on a scale of 15% so approx. 37x36 with a height of 15mm in Cura
4. loaded 3mf files and stl files with the same problem

### Actual results

the layers are incorrect, also with low layer heights
when I tried the same in Creality slicer with a bigger layer height (which should be worse) the problem didn't occur.

### Expected results

that the layers wouldn't be damaged, it also happened with other propellers; I don't know if it is only with propellers or also with other files.
At first, I thought the blades were maybe too thin, but scaling it bigger still caused it, and on Creality Slicer, there was nothing bad to see.

### Add your .zip and screenshots here ⬇️

[ezyzip.zip](https://github.com/user-attachments/files/17733575/ezyzip.zip)
[image]
[image]

GregValiant, 2024-11-13:
Thanks for the report.
Without a project file we can't tell what might be going on.  Please post a project file.

The model has errors in it and needs repair.
When a feature develops feather edges (like the trailing edges of the blades) they can pose problems.  I lowered the "Minimum Line Width" to 0.30 and the "Layer Height" to 0.10, and the jagged edges of the blades went away.
[image]

I dropped to a 0.3 nozzle and a 0.10 layer height and the repaired model looks OK.  This would be my preference.
[image]

Post a Project File so we can see how you have set this up.  Right now it looks like a settings problem rather than a bug.

CaptainBrick, 2024-11-13:
I dont have a .3 mm nozzle so thats not the case and a layer height of 0.1mm shouldn't make any difference because the quality is bad with layer heights of .06 .1 .15 and .2

CaptainBrick, 2024-11-13:
of course its more precise with a smaller nozzle but I don't have one

CaptainBrick, 2024-11-13:
what do you mean with a program file? what kind is it then? certainly not a .gcode or .stl

GregValiant, 2024-11-13:
A project file ("File | Save Project") is a special 3mf file that will contain all your Cura settings, the model, and your printer definition.  Once you have saved your project open the folder and right click on the new 3mf file.  Select the "Send To... | Compressed (zip) folder" option.  Post the zip folder here.

The edges of the propeller become "feather edges" with 0 thickness.
This is the model sliced in PrusaSlicer.  I've turned off the supports to make it easier to see the problem.  This is at .2 layer height and .4 line width.  Look at the gaps between the extrusions.  Trying to print this would just be a failure.
[image]

This is Creality slicer with the default settings of 0.2 LH and 0.4LW.  I see the same problem here.  The model features and size are such that a 0.4 nozzle simply can't get the job done unless you dial the settings way down.
[image]

This is a UM3 in UM Cura 5.9.0 beta.2 with 0.4 nozzle, 0.35 Line Width, and 0.28 Minimum Line Width and at 0.10 Layer Height.
This has a chance to be successful.
[image]

CaptainBrick, 2024-11-14:
my creality slicer looked much better much a bigger layer height so that's weird in according all you've said
especially because cura is not working that great

CaptainBrick, 2024-11-15:
wow that helps thanks but has the cylinder in the middle still the same diameter and is MS 3D builder a free app?

GregValiant, 2024-11-15:
MS 3D Builder was free in the Microsoft Store.  They keep changing availability.  It was originally bundled with Windows 10, then it was in the App store.  What they have done for a Windows 11 version I don't know.

It's not very intuitive but it is good at altering STL's (or making a new one based on primitives).  You wouldn't be able to design that propeller in it because the blades are too complicated.  Altering it isn't a problem.

In this case I...

- I opened the propeller STL in 3D Builder and moved the model to X0 Y0, Z7.5
- Copied and pasted the model into the scene so there were two identical models.
- Exactly placed the copy over the original
- Moved the copy up 0.20mm
- Merged the copy and the original.

The hub was the same diameter but would be 0.2mm too tall.  That can be taken off by subtracting another shape, or scaling down just the "Z" by 0.2mm.

CaptainBrick, 2024-11-15:
thank you really much for this idea, lll try it out myself

GregValiant, 2026-09-08:
I'll go ahead and close this.

closed 2026-09-08 by GregValiant