Ultimaker/Cura #19904: Cura constantly crashing with slice automatically

joezappie, 2024-11-13:
### Cura Version

5.8.1

### Operating System

Windows 11

### Printer

S1 Pro with Sonic pad

### Reproduction steps

I've been having regular cura crashes for the past several months. Every update I hope that some bug was fixed that stops it but unfortunately it hasn't. 

I believe this has to do with "Slice Automatically" being enabled and making a change that causes Cura to reslice many times in a short period. I believe it also has to do with being in preview mode.

I can guarantee Cura to crash by:
1. Enable Slice Automatically
2. Switch to preview mode
3. Add a filament change post processing script
4. Type 2-3 numbers into the "layers box"

It seems that input boxes trigger on input, not on change, so typing fast into these boxes causes Cura to reslice several times in a short time, probably overlapping. 

Cura just closes. I don't get any crash report. Looking in the `cura.log` in `%appdata%` I don't see much in terms of why it closed.

This happens with any model, as simple as a basic cube so I have not included a project file. I also have slowly removed every extension I use except moonraker, to see if an extension was causing the crash. It only happens when slice automatically is enabled. I never get a crash when manually slicing.

**Proposed solution:**
Debounce slicing. I think slice automatically should delay ~500 ms. If another slice request is made within that time, it defers it again. This would reduce the number of times Cura reslices, whether that's from typing in an input box, or an extension causing a reslice after it just happened. Maybe Cura already does this, but to me it doesn't seem like that's happening. 

### Actual results

Cura crashes when making changes

### Expected results

Cura shouldnt crash when making changes

### Add your .zip and screenshots here ⬇️

I deleted my cura.log, reopened Cura, did the above steps and here's the log file after it crashed:
https://pastebin.com/vF8EDwWs

GregValiant, 2024-11-13:
Thanks for the report.
There was just a discussion on the UM Forum about this behavior.  I would think that an "anti-bounce" wait of 1/2 second would be about right.  

This is halfway between a bug and an Improvement request so I'm going to add the "Improvement" label to this.

Someone from the Cura team will take a look.

joezappie, 2024-11-13:
Thanks for the reply - some updated info:

I've been playing around with this a bit more, and I do find that I can also get Cura to crash just by slowly toggling the `enable` checkbox under filament change, which kind of leads me to believe maybe its not caused by what I suggested above as that should only trigger the automatic slicing once. But maybe internally something with that single change is still causing Cura to reslice several times, I dont know.

That said, I've sat here toggling enable once a second about 30 times while in the "Prepare" tab and cannot get Cura to crash. Once I switch back to preview, it crashes within a few clicks, almost always on the first however. 

Heres an updated video showing cura crashes after slowly toggling the enabled checkbox:
https://www.youtube.com/watch?v=aZ4x3V5gWQk

I have no issues with automatic slicing turned on and dragging the part around causing Cura to reslice. It only seems to be in the post processing scripts that its crashing.

I still think a debounce on reslicing would be a good addition though, despite maybe not a solution to my current issue.

GregValiant, 2024-11-13:
Someone from the Cura team will take a look (@HellAholic might find this interesting?).  The team is busy getting ready for the release of 5.9.0 stable so it may be a bit before something can be done about this.

GregValiant, 2024-11-14:
My thought was that there isn't anything within a post-processor that would control how Cura picks up and emits a change to a setting.  Could it be a "QML" thing?
It would be impossible to know when the user has finished typing but a short "don't start/restart auto-slicing just yet" dwell might mitigate this behavior.

I've spent the last 20 minutes playing with this trying to force the crash.  It seems to be very intermittent.  I'm unable to force the crash using @joezappie's method.  I tried in 5.8.0, 5.8.1, and 5.9.0beta.2.  5.9.0 crashed once, but it doesn't seem to want to duplicate.

FabioM3, 2024-12-16:
I also have these problems, I think the file ...AppData\Roaming\cura\stderr.log have the answer.
[stderr.log](https://github.com/user-attachments/files/18151672/stderr.log)

Slashee-the-Cow, 2025-03-06:
@FabioM3 Even the regular Cura log has the answer (it says it's going to take a snapshot then it starts loading shaders and hard crashes so bad it doesn't leave an error in the log - *stderr.log* just confirms that it's happening.

The problem - at least I'm 99% sure in my case (which is exactly the same as @FabioM3's, other than the stack trace goes through a couple of different plugins) is a race condition.

- I add a simple object to the scene (like the most basic cube you can imagine, 8 verts, 12 faces) or a tetrahedron (4 verts, 4 faces) and >90% chance it's a straight CTD.
- I add a sphere with a few hundred faces, it's fine.
  - I can then add a cube, tetrahedron, whatever, it's fine.
  - If I can load a simple shape without it crashing, I can then add stuff to my heart's content.
  - I load something which won't crash, delete it (leaving an empty scene) and add a simple object and it's about as likely to crash as if adding the simple object straight after launching Cura.
  - **Slashee's conclusion:** It's definitely something to do with being the first object in the scene.
- I add a simple object to the scene, **don't** add the SliceableObjectModifier(), emit the sceneChanged signal for my object, put in a delay, *then* add the SliceableObjectModifier() and emit the signal again it **significantly lowers** the chance of a crash (down to about 50%).
  - I tested with delays up to about 2000ms before giving up.
  - The delay would likely be related to CPU speed so a bodge, not a fix.
  - The object doesn't actually appear in the viewport until after the timer has run, I'm guessing it's based on the function which adds it finishing.
### Slashee's conclusion:
It tries to auto slice. It calculates the geometry quicker than it can load the rendering pipeline for the thumbnail. It doesn't check to see if it's ready to render before it tries to create the thumbnail. Crash.

Slashee-the-Cow, 2025-03-09:
### Further notes:
I should probably have said *last message* that I'm working on a plugin. Previously I was using trimesh load an STL to create some MeshData, make a SceneNode and add that to scene. About the only way I could get it to slice successfully (if the first object was a simple model) was:
1. Disable auto slice,
2. Add node to scene.
3. Set a threading.Timer for a large delay (3 seconds) - It has to run a separate function because the model doesn't actually enter the scene until the method which creates it returns.
5. Jiggle it back and forth a little (in this case +X0.01 then -X0.01)
6. Slight delay (20ms)
8. Turn on auto slicing.
9. Slight delay (20ms)
10. Jiggle it again. Without the long delay, the jiggling did nothing.
- Without the **pre**-jiggle, auto slicing would turn back on but wouldn't try to slice it. I don't know if it needed to be reminded it was in the scene, or what.
- Without the **post**-jiggle, CuraEngine would immediately quit unexpectedly, saying there was nothing new to slice (which Cura interpreted as an engine crash). The post-jiggle forces it to immediately cancel and then restart the slice (now with a change registered to slice).

This results in a very poor UX, especially if you have the "Automatic Slicing Toggle" plugin installed:
 1. Plugin icon switches to "paused" icon,
 2. A few seconds later, icon switches back to active and slicing progress bar starts.
 3. Slicing progress bar immediately switches to "Unable to slice" warning (I'm guessing that's the post-jiggle)
 4. About a second later, automatically slices again. Fairly high success rate.

After testing using *File > Open File(s)* and it loading my most simple models fine, I figured I'd switch to using CuraApplication.readLocalFile(). This improved the some of the models I was testing with (to the point where some of them seem to work 100% of the time) but has also introduced problems when loading *other* simple models which *were* working fine to about an 80% failure rate.

Once again the problem is auto slicing either being already on or turned back on immediately after adding the object to the scene. Fortunately this one doesn't require the jiggling trickery to get working **so far**, just to use a Timer to run a function to set the auto slice preference back to true a a couple of seconds after it's finished messing with the shape (and that function returns).

I believe it also confirms it's a more serious problem than just for us people monkeying around if I'm using the API call that the UI uses when you open a file and still experiencing issues. I don't believe a fixed deferral would fully fix the problem, whatever is letting it try and render the thumbnail before it's ready needs to make sure it's ready before running.

Slashee-the-Cow, 2025-03-10:
Okay, even more curious: even when it doesn't crash trying to render the thumbnail, *it still can't render it.*
Here's something I was getting from 5.8.1 (was testing out installing a .curapackage version I created by hand:
```
2025-03-10 20:24:09,252 - DEBUG - [MainThread] StacksOfShapes.StacksOfShapes.log [59]: addShapeLoad() about to run readLocalFile, file_url = PyQt6.QtCore.QUrl('file:///C:/Users/slashee/AppData/Roaming/cura/5.8/plugins/StacksOfShapes/StacksOfShapes/models/prisms/prism_rhombic.stl'), _expected_title = Rhombic Prism, _expected_filename = prism_rhombic.stl
2025-03-10 20:24:09,253 - INFO - [MainThread] cura.CuraApplication.readLocalFile [2008]: Attempting to read file file:///C:/Users/slashee/AppData/Roaming/cura/5.8/plugins/StacksOfShapes/StacksOfShapes/models/prisms/prism_rhombic.stl
2025-03-10 20:24:09,256 - DEBUG - [JobQueueWorker [2]] UM.Mesh.MeshData.calculateNormalsFromVertices [561]: Calculating normals took 0.0 seconds
2025-03-10 20:24:09,256 - DEBUG - [JobQueueWorker [2]] STLReader.STLReader._read [68]: Loaded a mesh with 36 vertices
2025-03-10 20:24:09,258 - DEBUG - [JobQueueWorker [2]] UM.Mesh.MeshData.approximateConvexHull [505]: approximateConvexHull(target_count=1024) Calculating 3D convex hull took 0.0004999637603759766 seconds. 8 input vertices. 8 output vertices.
2025-03-10 20:24:09,259 - DEBUG - [JobQueueWorker [2]] UM.FileHandler.ReadFileJob.run [87]: Loading file took 0.0 seconds
2025-03-10 20:24:09,260 - DEBUG - [JobQueueWorker [2]] UM.Mesh.MeshData.approximateConvexHull [505]: approximateConvexHull(target_count=1024) Calculating 3D convex hull took 0.0005028247833251953 seconds. 8 input vertices. 8 output vertices.
2025-03-10 20:24:09,276 - DEBUG - [MainThread] UM.Operations.OperationStack.push [72]: <UM.Operations.AddSceneNodeOperation.AddSceneNodeOperation object at 0x000001CB61F18580>, took 2ms
2025-03-10 20:24:09,281 - DEBUG - [MainThread] UM.Operations.OperationStack.push [72]: GroupedOp.(#=2) RotateOp.(node=<CuraSceneNode object: 'prism_rhombic.stl'>,rot.=Q<0.0,0.0,0.0,w=4.0>) TranslateOp.(node=<CuraSceneNode object: 'prism_rhombic.stl'>,trans.=<0.000,0.000,0.000>), took 2ms
2025-03-10 20:24:09,283 - DEBUG - [MainThread] StacksOfShapes.StacksOfShapes.log [59]: _on_file_loaded saw C:/Users/slashee/AppData/Roaming/cura/5.8/plugins/StacksOfShapes/StacksOfShapes/models/prisms/prism_rhombic.stl
2025-03-10 20:24:09,283 - DEBUG - [MainThread] StacksOfShapes.StacksOfShapes.log [59]: Found child node that seems like what we want: <CuraSceneNode object: 'prism_rhombic.stl'>
2025-03-10 20:24:09,284 - DEBUG - [MainThread] UM.Mesh.MeshData.approximateConvexHull [505]: approximateConvexHull(target_count=1024) Calculating 3D convex hull took 0.00046539306640625 seconds. 8 input vertices. 8 output vertices.
2025-03-10 20:24:09,284 - DEBUG - [MainThread] StacksOfShapes.StacksOfShapes.log [59]: _on_file_loaded got w/d/h of 3.0 x 0.5773502588272095 x 1.0
2025-03-10 20:24:09,285 - DEBUG - [MainThread] StacksOfShapes.StacksOfShapes.log [59]: _on_file_loaded scaling shape by 6.666666666666667
2025-03-10 20:24:09,399 - DEBUG - [MainThread] UM.Operations.OperationStack.push [72]: PlatformPhysicsOp.(trans.=<0.000,2.833,0.000>), took 1ms
2025-03-10 20:24:09,603 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [61]: Loading [C:\Program Files\UltiMaker Cura 5.8.1\share\cura\resources\shaders\transparent_object.shader]...
2025-03-10 20:24:11,288 - INFO - [MainThread] CuraEngineBackend.CuraEngineBackend._createSnapshot [331]: Creating thumbnail image (just before slice)...
2025-03-10 20:24:11,289 - DEBUG - [MainThread] UM.Scene.Camera._preferencesChanged [255]: Changing perspective mode to perspective.
2025-03-10 20:24:11,290 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [61]: Loading [C:\Program Files\UltiMaker Cura 5.8.1\share\cura\resources\shaders\overhang.shader]...
2025-03-10 20:24:11,291 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [61]: Loading [C:\Program Files\UltiMaker Cura 5.8.1\share\cura\resources\shaders\transparent_object.shader]...
2025-03-10 20:24:11,292 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [61]: Loading [C:\Program Files\UltiMaker Cura 5.8.1\share\cura\resources\shaders\striped.shader]...
2025-03-10 20:24:11,374 - INFO - [MainThread] cura.Snapshot.snapshot [204]: Trying to get the snapshot again.
2025-03-10 20:24:11,386 - DEBUG - [MainThread] UM.Scene.Camera._preferencesChanged [255]: Changing perspective mode to perspective.
2025-03-10 20:24:11,387 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [61]: Loading [C:\Program Files\UltiMaker Cura 5.8.1\share\cura\resources\shaders\overhang.shader]...
2025-03-10 20:24:11,388 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [61]: Loading [C:\Program Files\UltiMaker Cura 5.8.1\share\cura\resources\shaders\transparent_object.shader]...
2025-03-10 20:24:11,389 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [61]: Loading [C:\Program Files\UltiMaker Cura 5.8.1\share\cura\resources\shaders\striped.shader]...
2025-03-10 20:24:11,476 - INFO - [MainThread] cura.Snapshot.snapshot [204]: Trying to get the snapshot again.
2025-03-10 20:24:11,480 - DEBUG - [MainThread] UM.Scene.Camera._preferencesChanged [255]: Changing perspective mode to perspective.
2025-03-10 20:24:11,481 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [61]: Loading [C:\Program Files\UltiMaker Cura 5.8.1\share\cura\resources\shaders\overhang.shader]...
[... cut ...]
2025-03-10 20:24:13,778 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._onBackendQuit [1043]: Backend finished slicing. Resetting process and socket.
```
Don't ask me why it seemed to slice twice. Auto slicing was only turned back on once. But consistently trying and failing to render a thumbnail... makes me question if it's a race condition or a deeper problem.

And if you're wondering what's so special about my rhombic prism... nothing that I'm aware of. I've attached it though. Zipped because for whatever reason I can't attach an STL.

[prism_rhombic.zip](https://github.com/user-attachments/files/19159661/prism_rhombic.zip)

HellAholic, 2025-03-10:
@Slashee-the-Cow The thumbnail capture has an issue on the qt side, to get around the issue it attempts to make a screenshot 10 times, if it fails to capture a screenshot after 10 times it will not include a screenshot in the package.

The error is caused by race condition of two functions, one is trying to update the container stacks (settings and such) and the other one is trying to read the container stacks. The latter fails since the container is being updated while it's trying to access it.

The post processing script view blocks your access to the main Cura window to prevent these type of errors specifically, but auto slice circles around that blockade. You can also trigger the error by changing a setting value but it's harder to reproduce.

Side note, the CuraEngineTiledInfill plugin is also causing fails in your case, so you're trying to debug an issue with a known bug already causing issues. That plugin was removed from marketplace.

Slashee-the-Cow, 2025-03-10:
@HellAholic 
> The error is caused by race condition of two functions, one is trying to update the container stacks (settings and such) and the other one is trying to read the container stacks. The latter fails since the container is being updated while it's trying to access it.

What triggers each of these functions? If first one is "object being added to scene" and second one is "tries to slice" that would explain why my plugin disabling auto slice before adding a shape to the scene and re-enabling it a few seconds later helps (but not necessarily doesn't completely get rid of) the problem.

To be clear, right now my plugin is doing the following things:
1. In `__init__`, plugin adds a handler for `CuraApplication.getInstance().fileCompleted`
2. User clicks button to add model. This runs "add model" function.
3. Function checks to see if auto slice is on, and if so, disables it (by setting `general/auto_slice` to False)
3. Function stores filename of model that's about to be loaded (we change the object's name later) and that it's waiting for a model to appear as instance variables.
4. Function uses `QUrl.fromLocalFile()` to get a QUrl from the relative path I'm using to store my models.
5. Function runs `CuraApplication.readLocalFile()` with aforementioned QUrl
6. Model loading function ends.
7. Handler for `fileCompleted` gets called.
8. Handler checks to see if plugin is waiting for a model to load and checks filename compared to model it's expecting. If either are False, it returns.
9. Handler iterates through `CuraApplication.getInstance().getController().getScene().getRoot().getChilderen()` until it finds a `node` with the model filename we loaded and stores in a variable.
10. Handler sleeps for half a second in case it takes the scene some time to catch up or something., I don't know. Figured it can't hurt.
11. Handler gets bounding box of node and calculates how much it has to scale to get to the preset size (from the plugin settings).
12. Handler runs `node.scale()` with Vector of scale amount in each direction.
13. Handler checks `node.getPosition()` and if too far from centre, uses a standard `UM.Operations.TranslateOperation` to move it to the centre.
14. Handler uses `node.setName()` to set its friendly name (which it gets from the plugin)
15. If auto slice was disabled before loading model, creates a `threading.Timer` to run a function which exclusively sets `general/auto_slice` back to True on a delay (I've tried half a second, one second, two seconds, three seconds, five seconds, longer seems to help but nothing completely prevents it).
16. Handler returns.
17. Function to re-enable auto slice runs after specified delay.
18. Auto slice gets re-enabled. Flip coin and it has about the same odds of CTD or working fine.

Umm... hopefully that's not too verbose 🤭. If you can suggest something better than my "delay and pray" bodge, I'm all ears.

FabioM3, 2025-03-14:
I tested with various models and no crashes anymore!
win10 exe.zip

Slashee-the-Cow, 2025-03-14:
Seems to be working well for me  (Windows 11 23H2 and openSUSE Tumbleweed) too.

wawanbreton, 2025-03-17:
Cool, that's good news !! Thanks for the feedback 🙂

Slashee-the-Cow, 2025-03-17:
Any idea if this is going to make it into 5.10? Still working on my plugin, need to know what version to have as the cutoff for the nuclear option (disables auto slicing while plugin window is open) since I've been unable to find a consistently successful workaround.

wawanbreton, 2025-03-17:
Code for the fix is ready for review and should be merged quite soon, so it will definitely be into 5.10

FabioM3, 2025-03-19:
I'm having a problem in the beta with this model: https://www.thingiverse.com/thing:2975429
In v5.9 it don't fail to slice, maybe is something related to this bugfix.

[export-20250319-093301.zip](https://github.com/user-attachments/files/19340916/export-20250319-093301.zip)

GregValiant, 2025-03-19:
That is a different problem and you should start a new report.  The model has problems and needs repair.

Slashee-the-Cow, 2025-03-20:
So for whatever it's worth (which for my purposes, is actually something) it seems to have been introduced in 5.7.0. In my testing (which basically involves trying to trigger the bug a bunch of times until either it crashes or I give up) 5.6 withstood me continually throwing stuff at it and 5.7.0 lasted only slightly longer than 5.9.0 before packing it in.

MariMakes, 2025-04-10:
This is resolved in the 5.10 release 🎉 , you can download the Cura version with the fix here: [https://github.com/Ultimaker/Cura/releases/tag/5.10.0](https://github.com/Ultimaker/Cura/releases/tag/5.10.0)

I'll close this issue since it's resolved. Thanks again, and please let us know if you run into any other issues 💪
I also saw that this issue was not correctly listed in the release notes as resolved, so I'll try to rectify this as well as we can. Sorry for that!

closed 2025-04-10 by MariMakes