Ultimaker/Cura #19905: Slicing error

Audyd1988, 2024-11-14:
### Cura Version

5.8.1

### Operating System

Windows 11

### Printer

Elegoo Neptune 4 Max

### Name abnormal settings

No straing setting that I know of sliced on my ender and cr-10

### Describe model location

Yes

### Describe your model

Yes, Yes and No

### Add your .zip here ⬇️

[EN4MAX_Catrina Statue.zip](https://github.com/user-attachments/files/17741487/EN4MAX_Catrina.Statue.zip)

GregValiant, 2024-11-14:
Thanks for the report.
I can duplicate this problem with the project file (in 5.8.0).  I moved the model on the Cura build plate and it sliced.
Having to move or rotate a model was a known bug in 5.6 and it appeared to have been fixed in 5.7.0.  This is the second time today I had a failed slice of a model, but moving the model fixed it.  It would appear that some change in 5.8.x has let this bug out of it's cage.
The Cura team will take a look.
This is in 5.8.0
[image]

Audyd1988, 2024-11-14:
Yes after I read the question on the report page I moved it a little and was able to slice also
Thanks for looking into this, not sure why it has been coming up but thank you again
Audra Dunn 

 
 

image.png (view on web)