Ultimaker/Cura #19911: stop printing

Deadjester70, 2024-11-15:
### Cura Version

5.8.0

### Operating System

Win10

### Printer

elegoo neptune

### Reproduction steps

loses connection software can't find printer although i can job the extruder up. restart computer and it finds it again.

### Actual results

finds it after restart

### Expected results

print

### Add your .zip and screenshots here ⬇️

[_OpenCase - A Modular ATX PC Case (No hardware req) - 6796887.zip](https://github.com/user-attachments/files/17774389/_OpenCase.-.A.Modular.ATX.PC.Case.No.hardware.req.-.6796887.zip)

GregValiant, 2024-11-15:
Thanks for the report.
When you say "loses connection" what does that mean?  How are you sending the Gcode to the printer?
We would need a Cura.log file and a project file.  The instructions are on the page you filled out for the report.

LilBub, 2024-11-15:
I'm wondering - if connected to a Windows PC,  if the USB port has gone into hibernation mode (Settings - System - Power & Battery - Power Mode in Windows 11)

GregValiant, 2024-11-15:
One of the reasons that no one wants to tackle fixing "USB Printing" is this sort of question/problem.
I suppose it depends on the operating system (and hardware) but if "going to sleep" means closing the port and powering it down, then that can cause a print to fail when the data stream stops.
Cura pokes the port every so often by sending M115 (ask for temperatures) as a "keep host alive" signal.  The printer may send automated responses over the USB (M113 setting) which is supposed to be the same sort of thing and tell the computer that the printer is still active on the port.  Many computers do not consider that to be enough of a reason not to take a nap.

- If the port is _closed_ by the computer then all USB activity will stop.  If you happened to be printing, then it will stop.
- If the port is _powered down_ by the computer but is left open then it may keep printing, but when the computer comes back to life and re-powers the port then the printer will notice the 5volt signal which many firmware flavors consider as a "hard reset" that will cause the printer processor to reboot and the print will fail at that point.  The Hard Reset is supposed to allow a user to reboot a locked up mainboard.

I wrote an app that is a front-end for my Ender 3 Pro (I modestly called it "Greg's Toolbox").  I do everything through the app.  I copy the gcode to the SD card, insert that in the printer, then from the laptop I re-initialize the SD card, ask the printer to list the files on the card, import them to a dropdown box in the app, and then select the one I want and tell the printer to print that file.  All the pre-heating and leveling is done through the app.  So my USB is hooked up pretty much 24/7 and the only problem I've ever had was when Microsoft sends out a late night Windows update which causes the computer to reboot which in turn causes the printer to reboot and the print fails.  Even though I'm not actually sending Gcode over the USB - if I have to print overnight then I disconnect the USB cord.

"Greg's Toolbox" and the "Print from SD Card" page.  The app will simultaneously control 10 printers on 10 different USB ports and allow individual control/tuning of each one.  It will slice, dice, and splice gcode files.  In addition, it has extensive tools for continuing a stopped print.  I mean why not?  I'm retired and have nothing better to do.  I'm a little upset that in spite of the app my hair continues to gray.  I'll keep working on that.
[image]

closed 2024-12-09 by github-actions[bot]