Ultimaker/Cura #19913: AppImage doesn't work

eugenialoli, 2024-11-15:
### Cura Version

5.8.1

### Operating System

Debian-Testing

### Printer

none

### Reproduction steps

When trying to load it:

eugenia@Debian-Testing:~/Downloads$  ./UltiMaker-Cura-5.8.1-linux-X64.AppImage 
Gtk-Message: 17:28:23.596: Failed to load module "appmenu-gtk-module"
Gtk-Message: 17:28:23.596: Failed to load module "appmenu-gtk-module"
qt.qpa.wayland: EGL not available

When trying to load it with environment variable (got this idea from another bug report here):

eugenia@Debian-Testing:~/Downloads$ WAYLAND_DISPLAY="" ./UltiMaker-Cura-5.8.1-linux-X64.AppImage 
Failed to create wl_display (Connection refused)
qt.qpa.plugin: Could not load the Qt platform plugin "wayland" in "" even though it was found.
Gtk-Message: 17:27:34.603: Failed to load module "appmenu-gtk-module"
Gtk-Message: 17:27:34.603: Failed to load module "appmenu-gtk-module"
qt.glx: qglx_findConfig: Failed to finding matching FBConfig for QSurfaceFormat(version 4.1, options QFlags<QSurfaceFormat::FormatOption>(), depthBufferSize -1, redBufferSize 1, greenBufferSize 1, blueBufferSize 1, alphaBufferSize -1, stencilBufferSize -1, samples -1, swapBehavior QSurfaceFormat::SingleBuffer, swapInterval 1, colorSpace QColorSpace(), profile  QSurfaceFormat::CoreProfile)
qt.glx: qglx_findConfig: Failed to finding matching FBConfig for QSurfaceFormat(version 4.1, options QFlags<QSurfaceFormat::FormatOption>(), depthBufferSize -1, redBufferSize 1, greenBufferSize 1, blueBufferSize 1, alphaBufferSize -1, stencilBufferSize -1, samples -1, swapBehavior QSurfaceFormat::SingleBuffer, swapInterval 1, colorSpace QColorSpace(), profile  QSurfaceFormat::CoreProfile)
Could not initialize GLX
Aborted

GLX is working btw:
Extended renderer info (GLX_MESA_query_renderer):
    Vendor: Intel (0x8086)
    Device: Mesa Intel(R) Arc(tm) A310 Graphics (DG2) (0x56a6)
    Version: 24.2.6
    Accelerated: yes
    Video memory: 4048MB
    Unified memory: no
    Preferred profile: core (0x1)
    Max core profile version: 4.6
    Max compat profile version: 4.6
    Max GLES1 profile version: 1.1
    Max GLES[23] profile version: 3.2

### Actual results

App doesn't load

### Expected results

to load

### Add your .zip and screenshots here ⬇️

none

hogend, 2024-11-15:
Same here on Debian Sid:
```
hogend@pandora:~/Downloads$ ./UltiMaker-Cura-5.8.1-linux-X64.AppImage
qt.glx: qglx_findConfig: Failed to finding matching FBConfig for QSurfaceFormat(version 4.1, options QFlags<QSurfaceFormat::FormatOption>(), depthBufferSize -1, redBufferSize 1, greenBufferSize 1, blueBufferSize 1, alphaBufferSize -1, stencilBufferSize -1, samples -1, swapBehavior QSurfaceFormat::SingleBuffer, swapInterval 1, colorSpace QColorSpace(), profile  QSurfaceFormat::CoreProfile)
qt.glx: qglx_findConfig: Failed to finding matching FBConfig for QSurfaceFormat(version 4.1, options QFlags<QSurfaceFormat::FormatOption>(), depthBufferSize -1, redBufferSize 1, greenBufferSize 1, blueBufferSize 1, alphaBufferSize -1, stencilBufferSize -1, samples -1, swapBehavior QSurfaceFormat::SingleBuffer, swapInterval 1, colorSpace QColorSpace(), profile  QSurfaceFormat::CoreProfile)
Could not initialize GLX
Aborted (core dumped)
hogend@pandora:~/Downloads$ glxinfo
name of display: :0
display: :0  screen: 0
direct rendering: Yes
server glx vendor string: SGI
server glx version string: 1.4
server glx extensions:
    GLX_ARB_context_flush_control, GLX_ARB_create_context, 
    GLX_ARB_create_context_no_error, GLX_ARB_create_context_profile, 
    GLX_ARB_create_context_robustness, GLX_ARB_fbconfig_float, 
    GLX_ARB_framebuffer_sRGB, GLX_ARB_multisample, 
    GLX_EXT_create_context_es2_profile, GLX_EXT_create_context_es_profile, 
    GLX_EXT_fbconfig_packed_float, GLX_EXT_framebuffer_sRGB, 
    GLX_EXT_get_drawable_type, GLX_EXT_libglvnd, GLX_EXT_no_config_context, 
    GLX_EXT_texture_from_pixmap, GLX_EXT_visual_info, GLX_EXT_visual_rating, 
    GLX_INTEL_swap_event, GLX_MESA_copy_sub_buffer, GLX_OML_swap_method, 
    GLX_SGIS_multisample, GLX_SGIX_fbconfig, GLX_SGIX_pbuffer, 
    GLX_SGIX_visual_select_group, GLX_SGI_make_current_read, 
    GLX_SGI_swap_control
client glx vendor string: Mesa Project and SGI
client glx version string: 1.4
client glx extensions:
    GLX_ARB_context_flush_control, GLX_ARB_create_context, 
    GLX_ARB_create_context_no_error, GLX_ARB_create_context_profile, 
    GLX_ARB_create_context_robustness, GLX_ARB_fbconfig_float, 
    GLX_ARB_framebuffer_sRGB, GLX_ARB_get_proc_address, GLX_ARB_multisample, 
    GLX_ATI_pixel_format_float, GLX_EXT_buffer_age, 
    GLX_EXT_create_context_es2_profile, GLX_EXT_create_context_es_profile, 
    GLX_EXT_fbconfig_packed_float, GLX_EXT_framebuffer_sRGB, 
    GLX_EXT_import_context, GLX_EXT_no_config_context, GLX_EXT_swap_control, 
    GLX_EXT_swap_control_tear, GLX_EXT_texture_from_pixmap, 
    GLX_EXT_visual_info, GLX_EXT_visual_rating, GLX_INTEL_swap_event, 
    GLX_MESA_copy_sub_buffer, GLX_MESA_gl_interop, GLX_MESA_query_renderer, 
    GLX_MESA_swap_control, GLX_NV_float_buffer, GLX_OML_sync_control, 
    GLX_SGIS_multisample, GLX_SGIX_fbconfig, GLX_SGIX_pbuffer, 
    GLX_SGIX_visual_select_group, GLX_SGI_make_current_read, 
    GLX_SGI_swap_control, GLX_SGI_video_sync
GLX version: 1.4
GLX extensions:
    GLX_ARB_context_flush_control, GLX_ARB_create_context, 
    GLX_ARB_create_context_no_error, GLX_ARB_create_context_profile, 
    GLX_ARB_create_context_robustness, GLX_ARB_fbconfig_float, 
    GLX_ARB_framebuffer_sRGB, GLX_ARB_get_proc_address, GLX_ARB_multisample, 
    GLX_EXT_buffer_age, GLX_EXT_create_context_es2_profile, 
    GLX_EXT_create_context_es_profile, GLX_EXT_fbconfig_packed_float, 
    GLX_EXT_framebuffer_sRGB, GLX_EXT_no_config_context, GLX_EXT_swap_control, 
    GLX_EXT_swap_control_tear, GLX_EXT_texture_from_pixmap, 
    GLX_EXT_visual_info, GLX_EXT_visual_rating, GLX_INTEL_swap_event, 
    GLX_MESA_copy_sub_buffer, GLX_MESA_gl_interop, GLX_MESA_query_renderer, 
    GLX_MESA_swap_control, GLX_OML_sync_control, GLX_SGIS_multisample, 
    GLX_SGIX_fbconfig, GLX_SGIX_pbuffer, GLX_SGIX_visual_select_group, 
    GLX_SGI_make_current_read, GLX_SGI_swap_control, GLX_SGI_video_sync
Extended renderer info (GLX_MESA_query_renderer):
    Vendor: AMD (0x1002)
    Device: AMD Radeon RX 7900 XT (radeonsi, navi31, LLVM 19.1.3, DRM 3.59, 6.11.7-amd64) (0x744c)
    Version: 24.2.7
    Accelerated: yes
    Video memory: 20480MB
    Unified memory: no
    Preferred profile: core (0x1)
    Max core profile version: 4.6
    Max compat profile version: 4.6
    Max GLES1 profile version: 1.1
    Max GLES[23] profile version: 3.2
Memory info (GL_ATI_meminfo):
    VBO free memory - total: 18046 MB, largest block: 18046 MB
    VBO free aux. memory - total: 31902 MB, largest block: 31902 MB
    Texture free memory - total: 18046 MB, largest block: 18046 MB
    Texture free aux. memory - total: 31902 MB, largest block: 31902 MB
    Renderbuffer free memory - total: 18046 MB, largest block: 18046 MB
    Renderbuffer free aux. memory - total: 31902 MB, largest block: 31902 MB
Memory info (GL_NVX_gpu_memory_info):
    Dedicated video memory: 20480 MB
    Total available memory: 52593 MB
    Currently available dedicated video memory: 18046 MB
OpenGL vendor string: AMD
OpenGL renderer string: AMD Radeon RX 7900 XT (radeonsi, navi31, LLVM 19.1.3, DRM 3.59, 6.11.7-amd64)
```

t-cc, 2024-11-15:
probably duplicated of #19837

Alfcyber, 2024-12-19:
Hello, maybe it could be related(?),
https://github.com/Ultimaker/CuraEngine/issues/2192

rephlex, 2026-01-02:
same as #18891 #19837 #19913 #20353 #20706 #20806 #21089 #20516

rephlex, 2026-10-25:
If you are on testing: my company got tired of AppImage roulette on our customer machines and now builds Cura as plain signed .debs per Debian release, testing included. It is just apt after a one-time key setup:

```
curl -fsSL https://mirror.plustech.de/plustech.gpg | gpg --dearmor -o /usr/share/keyrings/plustech.gpg
echo "deb [signed-by=/usr/share/keyrings/plustech.gpg] https://mirror.plustech.de/debian/ plustech-trixie main" > /etc/apt/sources.list.d/plustech.list
apt-get update && apt-get install ultimaker-cura
```

(Once trixie is superseded we build for the new suite too, the matrix is on the site.) https://mirror.plustech.de/ , sources published, community repo.

shadowFAQs, 2026-11-04:
Had this same issue, fixed it thanks to guidance from [Elod Pal Csirmaz's blog](https://www.epcsirmaz.co.uk/tech_blog/2026-03-appimage-recompile-cura.html). The `libz3-4` library file is missing from the Cura image. Here's how I fixed it:

1. Obtain the AppImage tool [here](https://github.com/AppImage/appimagetool/releases), which is itself an AppImage. You'll use this to re-pack Cura into an AppImage after unpacking it and adding the libz file.
2. Download the Bookworm version of libz3-4 [here](https://packages.debian.org/bookworm/libz3-4)
3. Unpack the downloaded .deb file: `ar x {.deb file}`
4. Decompress data: `unxz data.tar.xz`
5. Untar data.tar (`tar -x < data.tar`) and verify the library file is there at `usr/lib/x86_64-linux-gnu/libz3.so.4`
6. Unpack the Cura AppImage: `./{cura AppImage file} --appimage-extract`. Verify a `squashfs-root` directory appears.
7. `cp libz3.so.4` from your .deb file to `squashfs-root/usr/lib/x86_64-linux-gnu/`
8. Run the AppImage tool to button everything back up into a new Cura executable: `squashfs-root CuraFixed.AppImage`
9. If you encounter an error about icons, remove the extension from the `Icon=` line in `squashfs-root/com.ultimaker.cura.desktop`
10. Execute your new fixed AppImage

Happy slicing!