Ultimaker/Cura #19918: I have sliced this file before and it won't let me now?

gillyag, 2024-11-17:
### Cura Version

5.4.0

### Operating System

MACOS Sequoia 15.1

### Printer

SOVOL SV06

### Name abnormal settings

sliced 4 models before no problem. sliced this model before 

### Describe model location

it started to slice and then stopped when I raised it on the y 2 mm

### Describe your model

yes, printed before. it is a chained angel kneeling

### Add your .zip here ⬇️

[ChainedAngelStatueD.stl.zip](https://github.com/user-attachments/files/17791277/ChainedAngelStatueD.stl.zip)

GregValiant, 2024-11-17:
Thanks for the report.
Sometimes model errors can be overcome, but some are severe enough to give Cura problems.
[image]

There aren't many errors but they are enough to fool a slicer.
I'll mark this as a duplicate and close it.  Cura 5.4 is 7 versions out of date.  I suggest you upgrade to 5.8.1.  5.9.0 will be out in a week or so.

This is the un-repaired model in 5.8.1.  I never trust the slices of models that have errors. 
[image]

closed 2024-11-17 by GregValiant

gillyag, 2024-11-17:
Hi!Thank you very much! I didn’t design it, and like I said, I’ve printed it before, I didn’t know there was a problem in the water tightness of the model. I also thought Cura auto-updated so I will try to figure that out now. Sorry for the trouble.- Gillian

image.png (view on web)
image.png (view on web)