Ultimaker/Cura #19923: Slicing fails at default orienetation only.

Nishandh, 2024-11-18:
### Cura Version

5.8.1

### Operating System

Windows 11 Home Genuine

### Printer

CREALITY Ender-3 S1 Pro 

### Name abnormal settings

Nothing specific was shown in dialogue box

### Describe model location

YES

### Describe your model

Model : 791346-jeep-kit-card, available at :  https://www.printables.com/model/791346-jeep-kit-card

### Add your .zip here ⬇️

[CURA_Bug_Report_JeepKitCard.zip](https://github.com/user-attachments/files/17804226/CURA_Bug_Report_JeepKitCard.zip)

The slicing fails at default orientation of model on bed only. When its rotated, it does slice. The error, as well as its solution( rotate) is reproducable in the specific CURA installation.
**There where over 75 over rides, I would update this thread , after trying to slice it without any over rides.**
Thankyou.

GregValiant, 2024-11-18:
Thanks for the report.
The project would not slice for me either.  The model is "not watertight" and has surfacing errors (some faces are inside out) so that is likely a big part of that.

You have a lot of settings that won't work with this model.  
Having "Remove all holes" enabled means a lot of parts just turn into a solid block.
Printing "One at a Time" doesn't work since this is a single model.  
You have your print acceleration at 30mm/sec².  That's a snails pace.  The default for an Ender 3 is 1000.
There are no bridges in this model having that enabled does nothing but take time in the slice and possibly affects other things.

Many of Cura's settings are inter-connected.  When you enable a lot of them that simply aren't required, you create a lot of overhead for the program and increase the chances that there will be a conflict, or just a lousy slice.

The model needs repair.  I use MS 3D Builder for working with STL's.  It's surprisingly good at it.
Click on the "Super Quality" profile and discard all your changes.  Then, go through the settings and really look at each one and decide if it needs an adjustment.  Then slice after a change and inspect the preview to see if it was a worthwhile change.  A shotgun approach (enabling everything) rarely works.

Here is the repaired model sliced with your settings.  I'm sure this isn't what you want.
[image]

This is with just the "Super Quality" profile.  My printer is an Ender 3 Pro and this would print perfect.
[image]

I'll leave the bug label on this for now but models with errors often cause trouble for slicers.

GregValiant, 2024-11-20:
The problem with this model is the hood of the jeep.  It has a "star in a circle" inset and the pocket the arcs fit into are not watertight.  Some of the "model connectors" also have flipped normals and rotating the model can allow Cura to slice.

I'm going to remove the bug label and close this as it appears to be caused by problems with the model, and your settings.  I hope you understand.

Here I've used "Mesh Tools" to "Split the model into parts" and I've deleted two of the arcs of the circle around the star.  The red areas are surfaces that are wrong.
[image]

Here are more areas that are open and can't slice.
[image]

The jeep can be fixed and the "star and circle" raised so they have some definition.  This is with my settings and I've thrown the jeep body out.
[image]

closed 2024-11-20 by GregValiant