Ultimaker/Cura #19924: Crash when adding new Printer

dasds-cpu, 2024-11-18:
### Cura Version

5.8.1

### Operating System

Windows 11

### Printer

Felix TEc 4 Dual

### Reproduction steps

1. Clicked on Drop-Down Printer Menu 
2. Clicked "Add Printer"

### Actual results

The Application closes without an Error Message

### Expected results

The Menu to add a Printer should have opened.

### Add your .zip and screenshots here ⬇️

I dont have more Information

GregValiant, 2024-11-19:
Thanks for the report.
I can't duplicate this.  Doing the triage here - at any given time I probably have 20 printers installed.  I just haven't had this problem on my Win10 system.
Do you happen to have the "StartUp Optimizer" plugin installed?  If you do then un-install it and try again.  Part of what it does is to hide a lot of the printer definitions from Cura so they don't load.

fieldOfView, 2024-11-19:
>  Part of what it does is to hide a lot of the printer definitions from Cura so they don't load.

But what it doesn't do is let Cura close without an error message.

JustTheBast, 2024-12-14:
Same here, on my Windows 10 system.  Whether the "add printer" dialogue is called from the menu, or as part of the "first install onboarding" process, in either case, an empty dialogue box is opened, the program freezes for about a second, and then crashes without an error message.

I have tracked this backwards through the versions, and the last version that could still open the dialogue without crashing was 5.60.  Thus, as a workaround, one can install 5.60 alongside the latest version, and add one's printer there — as long as it existed in the list, at that time.