Ultimaker/Cura #19930: Cannot update from 5.2.1 to 5.9.0 without losing data

tripartist1, 2024-11-20:
### Cura Version

5.9.0

### Operating System

Windows 10

### Printer

Ender 3v2 Klipper

### Reproduction steps

1. Install 5.9.0
2. Open software

### Actual results

When opening I am greeted by what appears to be everything being correct, but missing a ton of profiles. Some are there, others are not. I really didnt want to manually add hundreds of settings for the several profiles that were missing. I saw some people had luck deleting the most recent config folder (the newest one that you just installed) so I did that but now I am greeted by the setup, and need to add my printer and everything. 

I have tried: 

- Using cura backups to backup my 5.2.1, adding a dummy printer to get past the setup screen on 5.9.0, restoring backup on 5.9.0, but after the software closes and reopens, its the same thing, like nothing changed. Same setup pops up, no printer added.
- Uninstalling all versions after 5.2.1 and deleting their config folders, then reinstalling 5.9.0. Same setup screen

I have stayed on 5.2.1 because every new version gives me trouble with updating. Why cant settings just transfer over correctly? There has to be a way to let me update and not start over with ALL my settings and profiles...

### Expected results

The user should not have to do anything and ALL settings, profiles, plugins, printers, etc should transfer over.

### Add your .zip and screenshots here ⬇️

na

tripartist1, 2024-11-20:
UPDATE: I have uninstalled all versions after 5.2.1 again and manually installed each major consecutive update since 5.2.1 (5.3, 5.4). They all import my machine and profile settings correctly up to 5.4. Installing any version after and including 5.5 does not migrate the settings from 5.4.

GregValiant, 2024-11-20:
Thanks for the report.
The current stable version is 5.8.1 (six versions ahead of 5.5).  Have you tried installing it to see if the problem still exists?
(5.5 and 5.6 had some slicing issues that were fixed in 5.7.0.)

tripartist1, 2024-11-21:
I was able to get my profiles back, but had to manually readd my machine, start/end gcode, offsets, etc. Something isnt carrying over from the machine settings portion. The 5.8.1 version made no difference, same issue with using cura backups and it not keeping my machine settings after completing the restore and restarting the software. This might need looking into.