Ultimaker/Cura #19934: Model takes over an hour to slice.

Ricecrispies76, 2024-11-20:
### Cura Version

5.8.1

### Operating System

Windows 10

### Printer

Modix Big 60

### Reproduction steps

There's a file I made in 2021 in Fusion 360 that I am re-printing now. Back then, the model took about a minute to slice. Now it takes over an hour. It does eventually slice, but it takes a looooong time. 

Note, Model was exported in Inch instead of Millimeter, so scale to 2540% for an actual size. Overall size should be ~256mm x ~222mm x ~63mm.

Steps: 
Load model
Scale up 2540%
Choose slicing settings
Start slicing
Wait 1+ hours for slicing to finish

[Top.zip](https://github.com/user-attachments/files/17834342/Top.zip)

[MV3BIG60_BlowerTop.zip](https://github.com/user-attachments/files/17834371/MV3BIG60_BlowerTop.zip)

### Actual results

Slicing takes over an hour.

### Expected results

Slicing takes 1 to 2 minutes, like it did when I ran this file in 2021.

### Add your .zip and screenshots here ⬇️

[cura.zip](https://github.com/user-attachments/files/17834391/cura.zip)
[MV3BIG60_BlowerTop.zip](https://github.com/user-attachments/files/17834396/MV3BIG60_BlowerTop.zip)

GregValiant, 2024-11-20:
Thanks for the report.
I sliced the project file.  My log says:
"Slicing took 482.34 seconds" so about 8 minutes.  The second try was 480 seconds.
The print time is "Print time (hr|min|s): 35h 8m 15s".
This is on a bare bones Win10 Pro laptop.

Looking at your log file it says:
"Slicing took 473.46 seconds" which is pretty close.
The print time is "Print time (hr|min|s): 35h 8m 58s".

Dropping the model in the Z so that all those holes are below the top of the build plate - the slice took just 9 seconds.

In regards to your slice, I don't have any idea where the time went or what Cura was doing for the rest of the hour (I had an employee like that once).  I'm unable to duplicate the problem.
I seem to recall one other model that seemed to take forever to slice (10 hours?).  It took so long that I got bored and went to bed.

I'll leave the bug label on this.  It might be system related somehow.  Do you happen to have any USB devices plugged in to the computer?  If the devices are configured as serial devices then Cura will spend time trying to communicate with them as if they were printers.  That can chew up some time.  If you disable "USB Printing" in the Marketplace you could check.

Ricecrispies76, 2024-11-20:
Thanks for taking a look! Interesting that the log says 8 minutes. It's definitely longer than that in real time. It could be system related. Task Manager says Cura is taking over 80% of my CPU resources for that entire time and the fans are ramped all the way up. CPU is an i5-10300H, so a little outdated, but not terrible for a laptop CPU.

I do have a USB plugged in for a wireless mouse. I've disconnected it, and reslicing now. 

This is the only file I have with this many holes, and the only one with this slicing slow down, so I figured they could be related. The GCode file it spits out is like 75MB. Cura is only using ~1gb of my 16gb of ram, so I don't think that's the bottleneck.

Ricecrispies76, 2024-11-20:
At 26 mins, it jumped to here 

[image]

Ricecrispies76, 2024-11-20:
[cura.zip](https://github.com/user-attachments/files/17835329/cura.zip)

Finished faster this time. Basically spot on 30 minutes in real time. I've attached the new Log file.

GregValiant, 2024-11-20:
A mouse or keyboard aren't problems.  Headphones and some other devices can be internally set up as serial devices.  Cura's USB Printing keeps sending "M105" and the device keeps sending back "What!!!".  That goes on and on.

From the log:
Slicing took 2023.8981087207794 seconds
So yeah, it took over 30 minutes to slice.

The Cura team will take a look.  Cura 5.5 and 5.6 had bugs regarding models like this that have a ton of walls on the build plate (all those holes).  That particular bug was fixed but the fix might be slowing things down.

Ricecrispies76, 2024-11-21:
Hi @HellAholic,

I checked my temps and they were hovering around 90C during the slice, which is high, but not extreme for a laptop. I also don't have these issues on other prints. They slice just fine. There's something about the number of holes on this model that is making everything throw a fit.

I also noticed an issue with the gcode/support material. When I have support turned off, the print estimate is 33 hours. There is only about 2-3hrs worth of support material being added, just supporting that little lip on the right side in the pictures. The print time goes all the way up to 54 hours with support turned on. If I cover the holes in the bottom surface with Support Blocker, it still takes forever to slice, but it drops the print estimate time back down to 37 hours. I think maybe it's identifying the holes as overhangs needing support?

LilBub, 2024-11-21:
Just wanted to chime in here - I've encountered a similar situation with Cura versions over the past year that will do this.  I get a 50% progress bar in slicing and in task manager, will see some very low CPU states.  Cura will seem to be hanging - maybe doing something?  But when I check the tasks, its doing something - but I don't get the dreaded "Not Responding" caption.

Ricecrispies76, 2024-11-22:
Appreciate you guys taking a look. I was hoping to not have to dig up the original cad file, but if I need to fix it and optimize it, so be it.

Ricecrispies76, 2024-12-21:
> I added a ticket for investigation, internal reference CURA-12343<https://ultimaker.atlassian.net/browse/CURA-12343>
>
> @Ricecrispies76<https://github.com/Ricecrispies76> Can we use this model for performance testing purposes?

Go for it. Sorry I'm a bit late on the response.