Ultimaker/Cura #19935: Cura 5.9.0 Beta 2 to Stable - Custom Definitions with Customer Platform/Platform Textures Not Loading

mzhu1113, 2024-11-20:
### Cura Version

5.9.0

### Operating System

W11

### Printer

NA

### Reproduction steps

NA

### Actual results

NA

### Expected results

NA

### Add your .zip and screenshots here ⬇️

Hello!

Not sure what's going on but I have custom printer definitions pointing to existing platform and platform texture files. These basically contain some instructions for my employees for specific machines.

These worked in Beta 2 of 5.9.0 but the stable release now just show the default platform grid. Was there a change?

Thanks!

Excerpt of the definition for one of our fleet machines:

`  "metadata":
  {
      "visible": true,
      "has_machine_quality": true,
      "has_materials": true,
      "has_variants": true,
      "preferred_quality_type": "Elegoo_giga_layer_06024",
      "preferred_variant_name": "0.60mm",
      "variants_name": "Giga Nozzle Size",
      "author": "ELEGOO",
      "machine_extruder_trains": { "0": "elegoo_giga_extruder_1" },
      "quality_definition": "elegoo_orangestorm_giga",
      "platform": "elegoo_platform_giga.obj",
      "platform_offset": [0, 0, 0],
      "platform_texture": "elegoo_platform_giga.png"
  },`

mzhu1113, 2024-11-20:
Yup it's there. 
So the textures have transparency so I'm assuming that's not going to work going forward. I'll get around to changing the textures and get back later in the day.

Thanks!

Marcos2BR, 2024-11-21:
my "platform": "hevo_platform.stl" also became invisible in version 5.9.
my mesh https://www.tinkercad.com/things/5mRE8Yl5Wmk-hevoplatformstl

[image]

Marcos2BR, 2024-11-21:
ps: the platform is there, bottom view :
[image]

mzhu1113, 2024-11-21:
@HellAholic 
Ok, I removed all transparency and the default grid still overrides the platform texture. Just in case I pulled the ankermake platform png you used from the share cura resources (nice kitten btw 🤪) and temporarily changed the texture to that and on the top I still only see the default grid but the below you can see the texture, just like Marcos shows... with some caveats.

[image]

I tried the regular Ultimaker theme just in case but no dice.

mzhu1113, 2024-11-21:
> @mzhu1113 I think you're missing a part in the definition then. Could you check to see if it shows up as expected if you add `"has_textured_buildplate": true,` in the medatada part of the printer definition? 

Adding that key worked. Much appreciated.
Curious, can't this just be set to true if platform_texture or platform keys exist? Not sure what use case would be satisfied by having has_textured_buildplate.

Thanks again!

Marcos2BR, 2024-11-22:
> @mzhu1113 I think you're missing a part in the definition then. Could you check to see if it shows up as expected if you add `"has_textured_buildplate": true,` in the medatada part of the printer definition? `"metadata": { "visible": true, "has_machine_quality": true, "has_materials": true, "has_variants": true, "preferred_quality_type": "Elegoo_giga_layer_06024", "preferred_variant_name": "0.60mm", "variants_name": "Giga Nozzle Size", "author": "ELEGOO", "machine_extruder_trains": { "0": "elegoo_giga_extruder_1" }, "quality_definition": "elegoo_orangestorm_giga", "platform": "elegoo_platform_giga.obj", "platform_offset": [0, 0, 0], "has_textured_buildplate": true, "platform_texture": "elegoo_platform_giga.png" },`
> 
> ![image](https://private-user-images.githubusercontent.com/28710690/388444674-31d261e3-cb5c-4602-ae60-fc37766ddad3.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzIzMTM0MTAsIm5iZiI6MTczMjMxMzExMCwicGF0aCI6Ii8yODcxMDY5MC8zODg0NDQ2NzQtMzFkMjYxZTMtY2I1Yy00NjAyLWFlNjAtZmMzNzc2NmRkYWQzLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDExMjIlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMTIyVDIyMDUxMFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTUzNGFjMGJmMmQ5YjAwMWE2NGZlNmFlYjMzODVhZGM0ZmQxMzE5MjFkM2I2YjNiZDNmMGZhZjNmZWZiMGQzYjUmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.-AjRzr4idCP1aN6YN4exBloYkLnDfjMhJGQ9m5r_VDI)

thanks, it worked after i added "has_textured_buildplate": true,
[image]

cetemadi, 2024-12-03:
I'm having the same issue as #19949. I've never used a bed texture, just the `"platform": "creality_ender3.3mf"`

Prior to 5.9 the platform design was visible from above, but after upgrading it's only visible from below. I've tried the steps above and even tried to add a .png texture but the issue persists. 

[image]

Here is an excerpt of my printer definition. I even tried changing the platform Z offset but no dice.

`{
    "version": 2,
    "name": "Creality Ender-5",
    "inherits": "creality_base",
    "metadata":
    {
        "visible": true,
        "platform": "creality_ender3.3mf",
        "platform_offset": [0, 10, 0], 
        "has_textured_buildplate": true, 
        "platform_texture": "ender3.png",
        "quality_definition": "creality_base"
    },`