Ultimaker/Cura #19936: Auto scaled model to 10000% of original size

DeiAlis, 2024-11-20:
### Cura Version

5.9.0

### Operating System

Windows 11 Home

### Printer

Creality CR10V2

### Reproduction steps

1: File -> Open file(s) -> <filename>.stl  prepared in Autodesk Maya ( set in mm. units
2: Scaling Object "Pop Up" -> Auto Scaled model to 10000% of original size ( and the object is very large )

### Actual results

1: To get the object to the real size I have to manually scale it not to 100% ( hundred ) but to 1000% ( thousand )
4: tried this with .obj file as well -> same issue
5: opened the .stl and .obj files in FreeCad to measure them they are the exact scale as outputted in Maya 

### Expected results

After the model is scaled to 1000%  ( thousand ) then its scale matches the real world scale and the original 3D software scale ( Maya )

### Add your .zip and screenshots here ⬇️

[image]
[Tap_001.zip](https://github.com/user-attachments/files/17837162/Tap_001.zip)

DeiAlis, 2024-11-20:
> image.png (view on web)
> Your model dimensions are x 1.5935 unit , y 1 unit, z 1.5935 unit, stl
> file does not come with its units attached. Opening the model you attached
> in Cura or other slicing or modeling software will show the same
> dimensions, slicer software assume you're using mm as the unit. If the
> model is less than a certain threshold in dimensions, it's scaled up
> automatically to make it visible.
> I'm struggling to figure out what's your expected behavior here.
>
> Here is an example stl of x 15.935 unit , y 10 unit, z 15.935 unit, Cura
> will not resize this model.
>
> Tap_001_updated.zip

Hello HellAholic

Thank you for your response

This is exactly the result I am expecting ( the way this updated model is
loading in Cura )
Since in Maya my units are set to millimeters and the model is exact I
expect Cura to load it properly ( like the updated model you sent )

On my end I am on the opinion that all is done properly ( but Cura is not
loading it OK )

Can you confirm that you scaled the model x10 to ensure the model loads
properly ?

Thank you

Nik

GregValiant, 2024-11-21:
@DeiAlis  I turned off "Scale extremely small models" and loaded the original file.
[image]

It is so small that it hit the "AutoScale" threshold so Cura scaled it up.

I agree this isn't a Cura problem so I'll go ahead and remove the bug label and close this.  You can still respond here if you need more information regarding this.
Let us know if you feel I'm being premature with the closure.

closed 2024-11-21 by GregValiant

DeiAlis, 2024-11-21:
Awesome, thanks for the update!