Ultimaker/Cura #19938: 3DBenchy Slicing Failure when disabling "Use Adaptive Layers" while tweaking print settings.

JGuar52, 2024-11-22:
### Cura Version

5.9.0

### Operating System

Windows 10 Pro, RX 6800

### Printer

Sovol SV06

### Name abnormal settings

Turning off "Use Adaptive Layers" causes slicing failure. Re-enabling it reverses the problem.

### Describe model location

Model is centered on build plate. Rotating 90 deg worked once, repeated attempts to slice continued to fail.

### Describe your model

3DBenchy. This is the first time I have encountered the error with the model, have printed successfully before. It is not watertight. Mesh Fix did not correct the problem.

### Add your .zip here ⬇️

[3DBenchy.zip](https://github.com/user-attachments/files/17857044/3DBenchy.zip)

JGuar52, 2024-11-22:
This occurred during a long session (1-2hrs) of slicing, printing, tweaking settings, and re-slicing within the same instance of Cura.

GregValiant, 2024-11-22:
Thanks for the report.
The model error doesn't appear to be a problem.  I repaired the model with MS 3D Builder and the slice failed.
I can duplicate the failure in 5.9.0 but it doesn't happen in 5.8.0.

The Cura team will take a look.

JGuar52, 2024-11-22:
Thank you for your advice on the initial layer settings, I will tune that down!