Ultimaker/Cura #19939: Hole Horizontal Expansion not applying consistently on holes through overhangs and slanted top layers.

tripartist1, 2024-11-22:
### Cura Version

5.9.0

### Operating System

Windows 10

### Printer

Ender 3v2, Klipper

### Reproduction steps

Apply hole horizontal expansion to a vertical hole which starts/ends at a position causing it to move through an overhang or angled top surface.

### Actual results

When the hole starts on an overhang, the overhang angle can cause the holes perimeters to not "be enough" for the horizontal expansion to detect it as a hole. This appears to be about 240 degrees. See the following images:
[image]
[image]

Towards the bottom of the hole where it pierces through the overhang, you can see the walls do not complete enough of the perimeter to be detected as a wall, so the offset is not being applied.

### Expected results

Hole expansion is applied to the entire length of the hole, regardless of the angle of the wall it is passing through.

### Add your .zip and screenshots here ⬇️

Will upload if requested.

fieldOfView, 2024-11-22:
Unfortunately, once the "hole" is opened up as it is in the top-most layer of the second image, it is no longer a hole but just part of the outer wall. CuraEngine only looks at the model one layer at a time so it has no inkling of these particular line segments somehow being connected to  to other line segments in other layers that are part of a "hole".

GregValiant, 2024-11-22:
There is the newish (is that a word?) setting "Hole Horizontal Expansion Max Diameter".  Setting it to a value equal to or greater than the diameter of the hole should fix the problem on that particular model.

This is with Max Diameter at the default "0.0" and shows the problem with the 6mm hole I've put in the block.  The lower portion of the hole is correct because there is no chamfer, but the upper portion shows the error.
[image]

With the Max Diameter at 6mm (my hole diameter) the problem goes away.
[image]

Try adjusting the "Hole Horizontal Expansion Max Diameter" and let us know if it fixes the problem.

closed 2024-12-07 by github-actions[bot]