Ultimaker/Cura #19940: Grabbed 5.9.0, model won't slice  (5.8.1 sliced it fine)

MoFoQ, 2024-11-22:
### Cura Version

5.9/0

### Operating System

Ubuntu

### Printer

Ender 3 Pro

### Describe your model

https://www.thingiverse.com/thing:6793493 (slices fine in 5.8.1)

### Add your .zip here ⬇️

Box.stl was opened (using ctrl-O and selecting it) then went immediately to "slice"
Cura gives a generic "slicing failed" error; nothing in the console that would explain it (since I invoke the AppImage from command line)
[image]
[image]

It would be nice to get a better/descriptive error message.

GregValiant, 2024-11-22:
Thanks for the report.
My printer is an Ender 3 Pro and using my profiles I don't have a problem slicing any of the files.
Load the problem file in Cura and use the "File | Save Project" command.  Open the folder and zip the 3mf project file.  Post the zip folder here and maybe there will be a clue as to what is happening.

MoFoQ, 2024-11-23:
[export-20241122-212158.3mf.zip](https://github.com/user-attachments/files/17878617/export-20241122-212158.3mf.zip)

ironically, github doesn't like .3mf files even though it's actually a zip file :laughing:

GregValiant, 2024-11-23:
Changing from the "Chep" profile to the "Standard" profile allowed it to slice for me.
Rotating the model also allowed it to slice with the CHEP profile.

I'm placing my money on "Alternate Extra Wall".

MoFoQ, 2024-11-23:
> Changing from the "Chep" profile to the "Standard" profile allowed it to slice for me. Rotating the model 30° also allowed it to slice with the CHEP profile.
> 
> I'm placing my money on "Alternate Extra Wall".

Bingo!
[image]

Fascinating since 5.8.1 seems to not have a problem with it :thinking: 

now I wonder why that is :thinking: :thinking: 
Could it be 5.8.1 just doesn't do alternate extra wall calculations (either by logical determination or because code bug)? :thinking: :thinking: :thinking: 

(on a sidenote, I already printed it via the output from 5.8.1 and with no supports and it turned out amazing)

GregValiant, 2024-11-23:
There was a problem with infill % dropping to 0 when walls were close together.  The fix for that was applied in 5.9.0 and this might be an "unexpected consequence".

@HellAholic may take a look.