Ultimaker/Cura #19942: Cannot connect Bambu Lab X1C to Cura via WiFi

nnobuddy, 2024-11-22:
### Cura Version

5.9.0

### Operating System

Windows 10

### Printer

Bambu Lab X1C

### Reproduction steps

1. Settings, Printer, Add Printer, Non UltiMaker printer, Add a networked printer, Add printer by IP
2. 192.168.68.105 (as shown in X1C Network)
3. Add

### Actual results

Message: Could not connect to device

### Expected results

Cura connected with X1C, just like in Bambu Studio or Orca.

### Add your .zip and screenshots here ⬇️

[image]
[image]
[image]
[image]

GregValiant, 2024-11-22:
Thanks for the report.
I believe network connectivity only works with UM printers.  The comment "Could not connect to device" may well mean that a handshake request was sent but that the printer didn't acknowledge as was expected.  It isn't possible for Ultimaker to go out and buy one of everyone else's printers to check for compatibility.

I'm going to remove the bug label and close this as you have a 3rd party printer.  You may find help over on Reddit or directly from Bambu.

closed 2024-11-22 by GregValiant

nnobuddy, 2024-11-25:
> Thanks for the report. I believe network connectivity only works with UM printers. The comment "Could not connect to device" may well mean that a handshake request was sent but that the printer [...]

Thx very much for your quick response. For now I've uninstalled Cura from my computer. I'll see whether I can get a response from Bambu Lab engineers. :)