Ultimaker/Cura #19943: Wait for Nozzle Heat Up worked in 5.8.1 but value doesn't carry over when upgrading to 5.9

LilBub, 2024-11-22:
### Cura Version

5.9

### Operating System

Windows 11

### Printer

Ultimaker 2+

### Reproduction steps

Slice model in 5.8.X and note you have Wait for Nozzle Heatup checked off.
Upgrade to 5.9
Start a print.
Quickly become alarmed your printer is printing once the bed reaches the target temperature.
Look at the Machine Settings and note the checked value didn't carry over.

Utter some color language, count to 10 then submit a ticket.

[image]

### Actual results

Described above

### Expected results

Values from previous version should carry over.

### Add your .zip and screenshots here ⬇️

[image]

LilBub, 2024-11-22:
Well even checked off, my print (via Octoprint ) is starting without heating the nozzle.

Cura broke something and now I can't use Octoprint on that machine in 5.9

LilBub, 2024-11-23:
[export-20241122-185810.zip](https://github.com/user-attachments/files/17877582/export-20241122-185810.zip)

GregValiant, 2024-11-23:
Thanks for the report.
This report is sort of in two parts.  The setting should have come into 5.9 from the previous version.  There have been a couple of reports of settings not acting right.
The criteria for "Wait for Nozzle Heatup" setting visibility is that the firmware cannot be "UltiGcode" which is the default.  When I add a second UM 2+ I must change the Gcode Flavor to Marlin to make "Wait for nozzle heatup" visible and when it does show up it is set to False.  So it appears that when a new UM 2+ is installed that the Firmware is UltiGcode and so "Wait for Nozzle Heatup" is hidden and the default value is "False".

Then there is the temperature problem once Wait for Nozzle Heatup is set to True.

When I generate a gcode from your project the line is in the temperature prepends.

;Generated with Cura_SteamEngine 5.9.0
M140 S80
M105
M190 S80
M104 S230
M105
**M109 S230**
M82 ;absolute extrusion mode

I'm not very familiar with the UM machines but that should work.  Someone from the Ultimaker team will take a look.

(As an aside - Torgeir on the UM Forum is pretty good with the UM2's.  You might want to give him a shout over there.  Maybe he can shed some light on the temperature problem.)

LilBub, 2024-11-23:
Good call on Torgeir
Also, the Help - Export Package exports to a 3MF.  Which Github won't allow me to attach.
I had to Zip it.  Can Cura Zip the Export Package in a future build?  Just saves a step - I know how to but a noob might struggle.

GregValiant, 2024-11-23:
I don't know about the zip thing.  Linux, Windows, and MAC are all different and then there is Chrome.
Windows is easy as it's just "Open Folder" in Cura and then right click on the file and "Send To" | "Compressed (zip) Folder".  There might be a way to add a call to the Windows function as a sub-menu in the Help menu.  You would still have a second step to select the file to add to the zip folder.  Sounds like six of one and a half dozen of the other to me.

massis87, 2025-02-28:
Adding to this: the M109 and M190 commands are still added to the gcode, but for some reason they are commented out so the print simply starts without heating:
The fact that the M190 for the bed is actually on the same line as the filament weight comment indicates to me someone messed up in the code.

```
;FLAVOR:Marlin
;TIME:12608
;Filament used: 28.8267m
;Layer height: 0.2
;MINX:51.338
;MINY:67.63
;MINZ:0.2
;MAXX:218.998
;MAXY:124.834
;MAXZ:38.8
;TARGET_MACHINE.NAME:VORON Trident 300
;Generated with Cura_SteamEngine 5.9.0
M82 ;absolute extrusion mode
;Nozzle diameter = 0.4
;Filament type = PETG
;Filament name = PETG - mine
;Filament weight = [88.05713671875]; M190 S85
; M109 S260
print_start EXTRUDER=260 BED=85 CHAMBER=25
M141 S25
G92 E0
G92 E0
G1 F1800 E-0.4
;LAYER_COUNT:194
;LAYER:0
M107
```