Ultimaker/Cura #19946: Drop Down Mesh Support Bug

camuehring, 2024-11-23:
### Cura Version

5.9.0

### Operating System

Windows 11

### Printer

Creality Ender-3 Neo

### Reproduction steps

1. Label Second Object as Support in "Per Model Settings"
2. Hit "Select Settings" Button
3. Scroll Down to bottom or near bottom and check box "Drop Down Mesh Support."
4. Nothing happens. Which is the problem.

### Actual results

Nothing Happens.

My custom made object serving as a support for another object is being given its own supports because of this setting which is making it unusable, and I cannot change the setting because of this bug.

### Expected results

A box should have appeared in the "Per Model Settings" that could be unchecked to stop supports being created for an object already labeled as a support.

### Add your .zip and screenshots here ⬇️

No picture to provide

camuehring, 2024-11-26:
[CFFFP_Skyrim Styled Sarcophagus Support.zip](https://github.com/user-attachments/files/17911517/CFFFP_Skyrim.Styled.Sarcophagus.Support.zip)
[image]
[image]

Here it is. I included two screenshots also, in case it is helpful.

GregValiant, 2024-11-26:
I can duplicate the problem.  I can't get "Drop Down Support Mesh" to show up or function either.  It just doesn't want to show up.
I went back to 4.13.1 and there were still problems with this.  I do remember using it so there must be an oler version that it worked in.
Someone from the Cura team will take a look.

BTW - When I get models like this that have a lip all the way around that will need support, I usually print it in two pieces.  If you alter the model by subtracting a couple of 2.2mm cylinders, you can make locating holes in the two pieces.  Drop in a piece of filament and super-glue the pieces together.
Here I've sunk one model into the build plate so only the top prints.  The other model has a support blocker configured as a cutting mesh so the top part doesn't print.  I subtracted two 2.2mm diameter cylinders as locating holes.  With a piece of filament in each one they make dandy locating holes so the parts go together just right.
[image]

camuehring, 2024-11-26:
Mmm. That would be easier, wouldn't it. Thanks.