Ultimaker/Cura #19947: option to append print time to file name

michaelshewitt, 2024-11-24:
### Is your feature request related to a problem?

...I have saved number of files ready to load and print but then cannot remember the estimated print time of each to allow me to efficiently schedule - say, choosing to commence printing an item when going to bed that will be finished early enough in the morning to allow me to start another print before leaving for work.

### Describe the solution you'd like

Preferences>General>Opening and saving files
A checkbox that will, when checked, append the estimated print time to the system-generated file name

### Describe alternatives you've considered

...I currently do this manually (when I remember to)

### Affected users and/or printers

All

Asterchades, 2024-11-24:
This can already be achieved with both the *Gcode Filename Format Plus* and *Printjob Naming* plugins on the Marketplace.

GregValiant, 2024-11-25:
This should work in either of those plugins.
[base_name] [print_time_hours]h[print_time_minutes]m
That would result in a suggested file name similar to:
Timer Shade 6h20m.gcode

@michaelshewitt does one of the plugins work for you?  Can I close this?

closed 2024-12-11 by github-actions[bot]