Ultimaker/Cura #19952: Brim should be along the part to print, not ghosted area!

classicrocker883, 2024-11-26:
### Cura Version

5.9.0

### Operating System

Win11

### Printer

Ender-3 S1

### Reproduction steps

1. import part
2. select brim for bed adhesion 
3. use support blocker
3a. make transparent block big enough to cut through half of model
4. per model settings
4a. modify settings for overlap. cutting mesh.
4b. make: wall line count, top layers, bottom layers, infill density all to be`0`
5. SLICE

### Actual results

brim appears around the ghosted part

### Expected results

brim should follow around actual parts or normal areas to be printed

### Add your .zip and screenshots here ⬇️

[image]

GregValiant, 2024-11-26:
Thanks for the report.
Without a project file we can't tell exactly what's going on.  Use the "File | Save Project" command and then zip the 3mf file and post the zip file here.

What it looks like is that the back side of the model is not flat.  The brim is going around the area that is in actual contact with the build plate.  Your blockers will keep the left and right out of the print, but the model shape is being used for the brim, and so it's an oval.
Look through the build plate from the bottom.  The "contact patch" will be cyan in color.

Here I've forced the behavior by rotating the model so it isn't flat on the build plate.
[image]

From the bottom you can see why that is.  The dark blue indicates the portion of the model that is below the build plate, the cyan stripe indicates the contact patch, and the red area indicates the model is above the build plate and would require supports.
[image]

If my assumptions are correct then this is "expected behavior" rather than a bug.  Post a project file.

classicrocker883, 2024-11-27:
[DazzlingMigelo-Rottis.zip](https://github.com/user-attachments/files/17932882/DazzlingMigelo-Rottis.zip)

GregValiant, 2024-11-27:
Much easier to visualize.  Thanks.
The bottom of the model has a radius.  The contact patch of the bottom ends where the radius starts and so a brim is going to be "under" the overhang of the radius.
[image]

When you add "cutting mesh" they don't change the actual model, they just tell Cura "don't print this area".
If you want the brim to touch the actual edge you would need to use an app like MS3D Builder and subtract those blocks from the model.
[image]

It's always best to give any slicer the "Model you want" rather than "This is close except for _______".  None of the slicers have the ability to actually "alter" a model.  Most are like Cura in that they can "have an effect" on the model which is all we often need.  That is why what I see here is "expected behavior" rather than a bug.

You are printing with PLA.  That model has enough surface area on the build plate that it shouldn't need a brim BUT the radius is large enough that it might like support.  You could try "Make overhang printable" (but make sure to turn it off when you are done with it).

This is "Make overhangs printable" with the brim turned off.  A chamfer has been added so it wouldn't need supports.
[image]

This is with Generate Supports enabled at 50° overhang angle.
[image]

I don't see a bug here.  This is the way Cura works for the model you have provided.

classicrocker883, 2025-01-02:
let me reiterate if you cannot understand what I meant earlier.

In the first example picture I provided, the brim isn't doing it's job because I have cut the model's sides. so, it does work without the model being manipulated at all, but the **FINAL** part's brim *should* be along the "new"/"manipulated" part - if you will.

the brim should be flat against the part that has been *cut* off, it should *not* continue along the old *ghosted* part which was pre-manipulated.

so the brim should go straight up and down on the left and right sides, not continue as an oval.

smartavionics, 2025-01-02:
Like this, perhaps?

[image]

classicrocker883, 2025-01-03:
yes i suppose... but for my part the brim does not fill in that area as it does for yours

smartavionics, 2025-01-03:
Yes, that's what it should do rather than what it does now. The required fix is trivial for the devs to do.

classicrocker883, 2025-04-03:
would someone like to point me to the code/file responsible for adding a brim? I'd like to take a look at it.

smartavionics, 2025-04-04:
This is the change I made to my Cura sources, don't know if it is applicable to Cura 5.

```
@@ -438,6 +438,14 @@ Polygons SliceDataStorage::getLayerOutlines(const LayerIndex layer_nr, const boo
                 {
                     continue;
                 }
+                if (mesh.settings.get<bool>("cutting_mesh") &&
+                    mesh.settings.get<int>("wall_line_count") == 0 &&
+                    mesh.settings.get<size_t>("top_layers") == 0 &&
+                    mesh.settings.get<size_t>("bottom_layers") == 0 &&
+                    mesh.settings.get<double>("infill_line_distance") == 0)
+                {
+                    continue;
+                }
                 const SliceLayer& layer = mesh.layers[layer_nr];
                 if (for_brim)
                 {
````

classicrocker883, 2025-04-04:
have you tested with this change and does it fix the issue? or did u need more info, like the model I used to slice to recreate the problem?

smartavionics, 2025-04-05:
It has received minimal testing. That image I posted on Jan 2nd shows the effect the changed code has.