Ultimaker/Cura #19953: Adhesion button showing off in shortcut menu when is switched on in advanced menu

foxy51, 2024-11-26:
### Cura Version

5.9

### Operating System

Windows 11

### Printer

Factor 4

### Reproduction steps

Switched on the adhesion in the advanced menu

### Actual results

Still indicates off in shortcut menu at the top of the prepare screen.

### Expected results

The adhesion icon should show on in the top Right Hand bar.

### Add your .zip and screenshots here ⬇️

[UMF4_GORDIANNO1V1.zip](https://github.com/user-attachments/files/17918127/UMF4_GORDIANNO1V1.zip)

GregValiant, 2024-11-26:
Thanks for the report.
It appears that the quick menu "Adhesion" button is not updating when "Skirt" is selected.  It works correctly for "None", "Brim", and "Raft".
Someone from the Cura team will take a look.

foxy51, 2024-11-26:
> You don't often get email from ***@***.******@***.***>. Learn why this is important<https://aka.ms/LearnAboutSenderIdentification>
>
> Skirt does not qualify as bed adhesion, if you read the setting description for the recommended view, it shows that bed adhesion refers to brim or raft when enabled, things that actually help with adhesion of the model to the build plate. Skirt's job is priming the nozzle. So as far as bed adhesion and the button are concerned, having a skirt means that you do not have a bed adhesion strategy. It is grouped with the bed adhesion for ease of use, so you don't have to disable skirt if you're printing with a raft or brim.
> image.png (view on web)<https://github.com/user-attachments/assets/4d6a2236-9309-4b1f-81e4-dcb99cde37bb>
>
>
> This email is from JD Sports Fashion plc or one of its subsidiaries ("JD Sports Fashion Group"). The contents of this email and any attachments are confidential and are intended solely for the use of the intended recipient. The information in this email may not be used, copied or disclosed by any person other than the intended recipient. If you are not the intended recipient, please contact JD Sports Fashion plc at ***@***.***, quoting the name of the sender and delete the message from your system. E-mails sent to and from the JD Sports Fashion Group may be monitored and read for legitimate business purposes. Emails cannot be guaranteed to be secure or error-free, and you should protect your systems. The JD Sports Fashion Group does not accept any liability arising from interception, corruption, loss or destruction of this e-mail, or if it arrives late or incomplete or with viruses. JD Sports Fashion plc - Registered in England No. 1888425. Registered Office: Hollinsbrook Way, Pilsworth, Bury, Lancashire, BL9 8RR.

Thanks for the prompt reply.
Kind regards M. Fox

GregValiant, 2024-11-26:
I guess I can go ahead and close this as it is expected behavior.  @foxy51 let me know if you think it is premature.

closed 2024-11-26 by GregValiant