Ultimaker/Cura #19955: se imprime en el aire

pablo2021ro, 2024-11-26:
### Cura Version

5.6.0

### Operating System

wuindos 11

### Printer

anacubic cobra neo

### Reproduction steps

pase el modelo ,sin cambiar nada del cura

### Actual results

que estaba la adherencia y no el modelo

### Expected results

que saliera el modelo completo

### Add your .zip and screenshots here ⬇️

[hopy hscp.zip](https://github.com/user-attachments/files/17918656/hopy.hscp.zip)

GregValiant, 2024-11-26:
Thanks for the report.
I think we're having a language problem here.  I'm not clear on the problem.  A few more words would be good.

Load the model and use the "File | Save Project" command.  Zip the 3mf file and post the zip folder here.  Looking at the gcode it seems like you did not have "Generate Supports" enabled.

pablo2021ro, 2024-11-26:
That is not the problem, the problem is that in the middle of printing it prints in the air and stops printing

GregValiant, 2024-11-26:
Were you printing via the SD card or using the USB, Octopprint, etc.?
Printing over the USB is can be very glitchy.
SD cards can have bad memory sectors and a print will start and run until the garbled code gets to the printer.  Reformatting the card will often fix that problem.
It is extremely unlikely that the problem is in the gcode.

pablo2021ro, 2024-11-27:
It tells me that the GCode file cannot be modified that ago

pablo2021ro, 2024-11-27:
You don't tell me it's a Hip code problem and I don't know what to do, what do I do next?

GregValiant, 2024-11-27:
When you load a gcode file in Cura, it's the toolpath representation of the model.  You can select "Print via USB" (if you have the printer connected and the computer and printer are actually talking to each other) or you can re-save the file, but there is no model in a gcode file and so there is no way to "re-slice" it.  A gcode file is just a text file and Cura is not a text editor.

You attached a gcode file.  Your printer is a "bed slinger" and that print will fail when it gets tall.  There just isn't enough contact with the build plate to keep the Y axis motion from breaking the print loose from the build plate.  That model needs support if for you other reason than build plate adhesion.
The first 20 or 30 layers of this print are printing over air.  It's going to fail because there is essentially zero adhesion and zero support.
You need to reload the model (STL or 3mf, etc.) and re-slice with supports turned on.

This is layer 1.  Failure is guaranteed.  Yes, it will be printing over air.
[image]

I don't see a bug here.  That gcode was sliced with someone's settings, and the settings were configured poorly.

pablo2021ro, 2024-11-27:
You could send me that file but it is edited so that it is with devices if it is active STL but if you send it to you it can be done please

pablo2021ro, 2024-11-27:
[image]

pablo2021ro, 2024-11-27:
I get the error and I don't know what to do. I've seen videos and I don't know what to do.

GregValiant, 2024-11-27:
That message is not an error.  Notice the yellow "!".  That indicates it is for "information" purpose only.  It is to remind you that gcode files are sliced for specific printers.  You can simply dismiss the message and move on.

Let's start over.
You load an STL file into Cura.  You make the settings so you like them and the preview looks good.
You slice, and then save the Gcode file to your hard drive.
You copy the Gcode file to an SD card.
Insert the SD card into the printer slot.
On the printer LCD you select the file you want to print.

A gcode file is text file.  It is a set of instructions for where the printer should move to, how hot it should be, how fast it should go.  It cannot be changed in Cura.  It is possible to print a gcode over the USB port from within Cura but USB Printing is not nearly as reliable as using the SD card.

pablo2021ro, 2024-11-27:
ok